canonical/ubuntu-frame #5: If two providers are connected to `graphics-core20` it doesn't work

AlanGriffiths, 2021-06-24:
```
2021-06-24T09:40:30Z ubuntu-frame.daemon[2627]:         # mode: 1920x1080@60.0    # Defaults to preferred mode
2021-06-24T09:40:30Z ubuntu-frame.daemon[2627]:         # position: [0, 0]    # Defaults to [0, 0]
2021-06-24T09:40:30Z ubuntu-frame.daemon[2627]:         # orientation: normal    # {normal, left, right, inverted}, defaults to normal
2021-06-24T09:40:30Z ubuntu-frame.daemon[2627]:         # scale: 1
2021-06-24T09:40:30Z ubuntu-frame.daemon[2627]:         # group: 0    # Outputs with the same non-zero value are treated as a single display
2021-06-24T09:40:30Z ubuntu-frame.daemon[2627]: 8>< ---------------------------------------------------
2021-06-24T09:40:30Z ubuntu-frame.daemon[2627]: [2021-06-24 09:40:30.569830] < -warning- > gbm-kms: drmModeCrtcSetGamma failed: Invalid argument
2021-06-24T09:40:30Z ubuntu-frame.daemon[2627]: Mir fatal error: Failed to perform buffer swap
2021-06-24T09:40:32Z systemd[1]: snap.ubuntu-frame.daemon.service: Main process exited, code=dumped, status=6/ABRT
2021-06-24T09:40:32Z systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'core-dump'.

10:41
ubuntu@mtk-aiot:~$ snap connections 
Interface                 Plug                          Slot                                Notes
content[graphics-core20]  ubuntu-frame:graphics-core20  mesa-core20:graphics-core20         -
content[graphics-core20]  ubuntu-frame:graphics-core20  mtk-drivers-core20:graphics-core20  manual
opengl                    ubuntu-frame:opengl           :opengl                             -
x11                       ubuntu-frame:x11              :x11                                -
```

AlanGriffiths, 2021-06-24:
> When multiple slots are connected to the same plug and they share directories with the same name, those directories are given unique names with the following pattern: \<directory>, \<directory>-2, \<directory>-3, \<directory>-x. The names of shared directories with unique names are retained, as defined by the slot.

So we're just seeing the first slot provider and ignoring subsequent ones. There are two options:
1. error if there are multiple slot providers; or,
2. detect multiple slot providers and use them all.

The latter is most flexible and probably feasible: all three related environment variables are colon-separated list of paths. But that doesn't prevent bad interactions between different providers. These could be hard to diagnose.

@RAOF you've most experience in this area, WDYT?

Saviq, 2021-06-24:
Does glvnd play into this at all?

AlanGriffiths, 2021-06-24:
> Does glvnd play into this at all?

It could be involved but, of itself, that's mostly harmless.

Saviq, 2021-06-24:
> It could be involved but, of itself, that's mostly harmless.

42.

RAOF, 2021-06-24:
> So we're just seeing the first slot provider and ignoring subsequent ones. There are two options:
>
>     1. error if there are multiple slot providers; or,
>
>     2. detect multiple slot providers and [...]

In the absence of something like glvnd there will be only one EGL implementation that gets linked in, and the others will be ignored - or at least should be.

For drivers that do not use glvnd then there is no point in having multiple providers connected; only one can possibly work.

For drivers which *do* support glvnd we can probably detect them via the existence of one or more files in `/usr/share/glvnd/egl_vendor.d`. We'd need to add the various paths to the [relevant environment variable](https://github.com/NVIDIA/libglvnd/blob/master/src/EGL/icd_enumeration.md), which would be `__EGL_VENDOR_LIBRARY_DIRS`.

AlanGriffiths, 2021-06-24:
Thanks @RAOF! So it should be possible, but there's bunch of stuff to orchestrate to make it "just work".

Given that, pending a usecase that justifies doing the work, I think the way ahead is to detect multiple providers being connected and give a meaningful error.

Saviq, 2023-05-03:
Fixed by #8

closed 2023-05-03 by Saviq