canonical/ubuntu-frame #19: fullscreen vs maximize

wmww, 2021-09-10:
Right now we set toplevel windows to be fullscreen, but this doesn't propagate to clients. This can be observed by running gedit. In fullscreen mode (press F11) it has no titlebar, but in ubuntu-frame it does have one.

Fixing this is further complicated when introducing the OSK. We want apps to *think* they're fullscreen (maybe?), but be treated by the window manger like they're maximized (so they move out of the way of the OSK). In other words, the exact oposite of the current behavior.

wmww, 2021-09-10:
Related to https://github.com/MirServer/mir/issues/2167

wmww, 2021-09-16:
It appears the real problem is that gedit was initially maximized, and for some reason Ubuntu Frame only fullscreens surfaces that are initially restored.

AlanGriffiths, 2021-11-11:
Fixed by #21

closed 2021-11-11 by AlanGriffiths