canonical/ubuntu-frame #20: Difficult to bootstrap the `display` config option

Saviq, 2021-09-14:
It's currently difficult to retrieve the defaults for `frame.display`. We point at `/var/snap/ubuntu-frame/current/frame.display`, but that's an internal detail.

It would be good to `snapctl set display …` with the default config, since that what ends up in there anyway?

AlanGriffiths, 2021-09-20:
Once the display layout is set we don't handle subsequent changes to the physical outputs very well. For now, I'd rather we "opt in" by manually setting a configuration. But I'm very open to ideas for improving this.

Saviq, 2021-09-21:
Well, since the file is written by ubuntu-frame anyway, it's not really opt-in, is it?

AlanGriffiths, 2021-09-21:
Well, Mir won't write that file until it runs as a daemon for the first time. As an example, I don't have it on my system.

Saviq, 2021-09-21:
Sure, but on Ubuntu Core, just installing `ubuntu-frame` makes the current setting the persistent one. But it's still not available through `snap get` until you manually `snap set` it.

AlanGriffiths, 2021-09-21:
Right, but when do you envisage setting it?

There's file until after Mir has run as a daemon for the first time, so dumping `snapctl set display="$(cat ${display_file})"` in `hooks/configure` would not take effect until a subsequent "configure".

We probably want Mir to do some better change management (e.g. adding any new devices it encounters). And maybe have a "dump out the display config and exit" option too.

AlanGriffiths, 2021-09-21:
Ah! Yeah, it is a bit weird moving it out of the way only in the configure hook.

You'd be happy if we move the `mv "${display_file}"  "${display_file}.save" || true` incantation from `configure` to `run-frame`? Want to PR that?

Saviq, 2021-09-21:
But isn't `${display_file}` used when you `snap set ...`, too? That doesn't sound right to move it about?

AlanGriffiths, 2021-09-21:
Yes, if `$(snapctl get display)` returns something (in `configure`) we write the file.

At present, if `$(snapctl get display)` returns nothing (in `configure`) we move the file out of the way.

Hypothetically, "if `$(snapctl get display)` returns nothing (in `run-frame`) we move the file" would give behaviour you're happy with?

commit f0e45654 (2023-01-27): Automate reloading display configuration (#122)

This allows Frame to handle display reconfiguration without a restart.

Features:

1. If there is no `display` option set, then when Frame runs it is set to a default
2. When the `display` option is set:
2.1. With Miral < 3.8 Frame is restarted and will load the new display configuration
2.2. With Miral 3.8 Frame is not restarted and will attempt to reload
3. When the display option is unset, the configuration is cleared and Frame restarted

Note: there is (still) no validation of the configuration being set

Fixes: #20
closed 2023-01-27 by bors[bot]