canonical/ubuntu-frame #39: Fail to start Web Kiosk example

flaprimo, 2021-11-11:
I've followed the steps described [here](https://github.com/MirServer/ubuntu-frame/blob/master/RUNNING_ON_YOUR_DEVICE.md) to set up a raspberry pi with Ubuntu Server 21.10 to show a web page with no success.

Steps:

1. Flash the Ubuntu Server 21.10 on an sdcard and insert into the Raspberry Pi
2. Connected the Raspberry Pi to an HDMI external monitor
3. Follow the [instructions](https://github.com/MirServer/ubuntu-frame/blob/master/RUNNING_ON_YOUR_DEVICE.md) via an SSH connection. After running `snap set ubuntu-frame daemon=true` command the external monitor turns black and the cursor starts blinking on the top left of the screen. Even after entering the remaining commands the screen remains the same without showing a web page.

Thank you in advance.

AlanGriffiths, 2021-11-11:
Hi, can you supply a bit more information?

1. which model RPi are you using?
2. could you pastebin the logs?

```
snap logs -n 100 ubuntu-frame
```

flaprimo, 2021-11-11:
1. Raspberry Pi version: Raspberry Pi 3 Model B Plus Rev 1.3
2. There you go for the logs:
```
ubuntu@ubuntu:~$ sudo snap logs -n 100 ubuntu-frame
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: [2021-11-11 09:23:50.797064] < - debug - > mirserver: Using Linux VT subsystem for session management
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: [2021-11-11 09:23:50.798013] <information> mircommon: Loading modules from: /snap/ubuntu-frame/488/usr/lib/aarch64-linux-gnu/mir/server-platform
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: [2021-11-11 09:23:50.798522] <information> mircommon: Loading module: /snap/ubuntu-frame/488/usr/lib/aarch64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.19
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: [2021-11-11 09:23:50.798733] <information> mircommon: Loading module: /snap/ubuntu-frame/488/usr/lib/aarch64-linux-gnu/mir/server-platform/server-x11.so.19
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: [2021-11-11 09:23:50.798819] <information> mircommon: Loading module: /snap/ubuntu-frame/488/usr/lib/aarch64-linux-gnu/mir/server-platform/input-evdev.so.8
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: [2021-11-11 09:23:50.800753] <information> gbm-kms: Unsupported: No DRM devices detected
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: [2021-11-11 09:23:50.800888] <information> mirserver: Found graphics driver: mir:gbm-kms (version 2.5.0) Support priority: 0
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: [2021-11-11 09:23:50.801049] <information> mirserver: Found graphics driver: mir:x11 (version 2.5.0) Support priority: 0
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: ERROR: /build/mir-lz9tNI/mir-2.5.0/src/server/graphics/default_configuration.cpp(231): Throw in function virtual const std::vector<std::shared_ptr<mir::graphics::DisplayPlatform> >& mir::DefaultServerConfiguration::the_display_platforms()
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: Dynamic exception type: boost::wrapexcept<std::runtime_error>
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: std::exception::what: Exception while creating graphics platform
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: ERROR: /build/mir-lz9tNI/mir-2.5.0/src/server/graphics/platform_probe.cpp(137): Throw in function std::vector<std::shared_ptr<mir::SharedLibrary> > {anonymous}::modules_for_device({anonymous}::ModuleType, const std::vector<std::shared_ptr<mir::SharedLibrary> >&, const mir::options::ProgramOption&, const std::shared_ptr<mir::ConsoleServices>&)
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: Dynamic exception type: boost::wrapexcept<std::runtime_error>
2021-11-11T09:23:50Z ubuntu-frame.daemon[17135]: std::exception::what: Failed to find any platforms for current system
2021-11-11T09:23:50Z systemd[1]: snap.ubuntu-frame.daemon.service: Main process exited, code=exited, status=1/FAILURE
2021-11-11T09:23:50Z systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'exit-code'.
2021-11-11T09:23:53Z systemd[1]: snap.ubuntu-frame.daemon.service: Scheduled restart job, restart counter is at 236.
2021-11-11T09:23:53Z systemd[1]: Stopped Service for snap application ubuntu-frame.daemon.
2021-11-11T09:23:54Z systemd[1]: Started Service for snap application ubuntu-frame.daemon.
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: + exec nohup /snap/ubuntu-frame/488/bin/run-frame /snap/ubuntu-frame/488/usr/local/bin/frame
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: + [ /snap/ubuntu-frame/488/usr/local/bin/frame = --help ]
2021-11-11T09:23:54Z ubuntu-frame.daemon[17180]: + dirname /run/user/0/snap.ubuntu-frame
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: + XDG_RUNTIME_DIR=/run/user/0
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: + export XDG_RUNTIME_DIR
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: + mkdir -p /run/user/0 -m 700
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: + [ -n  ]
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: + exec /snap/ubuntu-frame/488/usr/local/bin/frame
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: [2021-11-11 09:23:54.296156] <information> mirserver: Starting
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: [2021-11-11 09:23:54.298098] < - debug - > mirserver: Using Linux VT subsystem for session management
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: [2021-11-11 09:23:54.299152] <information> mircommon: Loading modules from: /snap/ubuntu-frame/488/usr/lib/aarch64-linux-gnu/mir/server-platform
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: [2021-11-11 09:23:54.299633] <information> mircommon: Loading module: /snap/ubuntu-frame/488/usr/lib/aarch64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.19
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: [2021-11-11 09:23:54.299878] <information> mircommon: Loading module: /snap/ubuntu-frame/488/usr/lib/aarch64-linux-gnu/mir/server-platform/server-x11.so.19
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: [2021-11-11 09:23:54.300001] <information> mircommon: Loading module: /snap/ubuntu-frame/488/usr/lib/aarch64-linux-gnu/mir/server-platform/input-evdev.so.8
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: [2021-11-11 09:23:54.302000] <information> gbm-kms: Unsupported: No DRM devices detected
2021-11-11T09:23:54Z ubuntu-frame.daemon[17162]: [2021-11-11 09:23:54.302189] <information> mirserver: Found graphics driver: mir:gbm-kms (version 2.5.0) Support priority: 0
[... cut ...] n&, const std::shared_ptr<mir::ConsoleServices>&)
2021-11-11T09:24:01Z ubuntu-frame.daemon[17212]: Dynamic exception type: boost::wrapexcept<std::runtime_error>
2021-11-11T09:24:01Z ubuntu-frame.daemon[17212]: std::exception::what: Failed to find any platforms for current system
2021-11-11T09:24:01Z systemd[1]: snap.ubuntu-frame.daemon.service: Main process exited, code=exited, status=1/FAILURE
2021-11-11T09:24:01Z systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'exit-code'.
```

AlanGriffiths, 2021-11-11:
From the Ubuntu Frame POV the problem is this:
```
2021-11-11T09:24:01Z ubuntu-frame.daemon[17212]: [2021-11-11 09:24:01.825097] <information> gbm-kms: Unsupported: No DRM devices detected
```
That is, the kernel isn't exposing any DRM devices.

As to how to fix that, I don't immediately know why there would be no DRM devices on an Ubuntu Server 21.10 install. What I do know is that the RPi3 graphics hardware has needed specific kernel configuration in the past. For example:

https://discourse.ubuntu.com/t/mir-kiosk-on-a-raspberry-pi3-with-ubuntu-server-18-04/12918/

The [Impish Release notes](https://discourse.ubuntu.com/t/impish-indri-release-notes/21951) also mention `dtoverlay`, but until I find time to experiment (may be a few days) I don't know if or how that affects Mir.

flaprimo, 2021-11-11:
In [Impish Release notes](https://discourse.ubuntu.com/t/impish-indri-release-notes/21951) I've followed the suggestion to change the `dtoverlay` setting in `config.txt` (from `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d`).
In `config.txt` I found `dtoverlay=dwc2` and changed it to `dtoverlay=vc4-fkms-v3d`, and now it displays `wpe-webkit-mir-kiosk`.

Only prolem is that now I get the error mentioned [here](https://gitlab.com/glancr/wpe-webkit-snap/-/issues/27) regarding `TLS Error: TLS certificate is expired`.

tobias-grasse, 2021-11-12:
@flaprimo
* Are you setting https://mir-server.io as URL?
* Which revision of `wpe-webkit-mir-kiosk` is installed? Check with `snap info wpe-webkit-mir-kiosk | tail -n1`
* I guess you are on the armhf channel, but to be sure, please check that you're not on the arm64 channel: 
`snap run --shell wpe-webkit-mir-kiosk.restart-watcher`
Inside the snap sheep, run `echo $SNAP_ARCH` → should be armhf