canonical/ubuntu-frame #47: Virtualbox support

EinarArnason, 2021-12-03:
Are there plans to support Virtualbox?

When I try to run in VM I get this message using the VMSVGA graphics controller:
`
[2021-12-03 12:39:56.918150] < - debug - > mirserver: Using Linux VT subsystem for session management
[2021-12-03 12:39:56.918244] <information> mircommon: Loading modules from: /snap/ubuntu-frame/489/usr/lib/x86_64-linux-gnu/mir/server-platform
[2021-12-03 12:39:56.918275] <information> mircommon: Loading module: /snap/ubuntu-frame/489/usr/lib/x86_64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.19
[2021-12-03 12:39:56.918285] <information> mircommon: Loading module: /snap/ubuntu-frame/489/usr/lib/x86_64-linux-gnu/mir/server-platform/server-x11.so.19
[2021-12-03 12:39:56.918293] <information> mircommon: Loading module: /snap/ubuntu-frame/489/usr/lib/x86_64-linux-gnu/mir/server-platform/input-evdev.so.8
[2021-12-03 12:39:56.919082] <information> gbm-kms: Unsupported: No DRM devices detected
[2021-12-03 12:39:56.919106] <information> mirserver: Found graphics driver: mir:gbm-kms (version 2.5.0) Support priority: 0
[2021-12-03 12:39:56.919127] <information> mirserver: Found graphics driver: mir:x11 (version 2.5.0) Support priority: 0
ERROR: /build/mir-mreCGI/mir-2.5.0/src/server/graphics/default_configuration.cpp(231): Throw in function virtual const std::vector<std::shared_ptr<mir::graphics::DisplayPlatform> >& mir::DefaultServerConfiguration::the_display_platforms()
Dynamic exception type: boost::wrapexcept<std::runtime_error>
std::exception::what: Exception while creating graphics platform
ERROR: /build/mir-mreCGI/mir-2.5.0/src/server/graphics/platform_probe.cpp(137): Throw in function std::vector<std::shared_ptr<mir::SharedLibrary> > {anonymous}::modules_for_device({anonymous}::ModuleType, const std::vector<std::shared_ptr<mir::SharedLibrary> >&, const mir::options::ProgramOption&, const std::shared_ptr<mir::ConsoleServices>&)
`

AlanGriffiths, 2021-12-03:
Mir (and Ubuntu Frame) does work in a VM with suitable graphics support. For an example using qemu/kvm see:

https://github.com/MirServer/ubuntu-frame/blob/master/RUNNING_IN_A_VM.md

I've used Mir with Virtualbox as recently as two years ago, but no longer have that system to hand. Without an instance to experiment with, I don't know how to go about configuring Virtualbox with the necessary with DRM/KMS support. Maybe switching on "3D" graphics?

EinarArnason, 2021-12-06:
I tried enabling all the graphic related features and ensured the vboxvideo driver is loaded, I still get the "No DRM devices detected" error. Could I be missing a package that's needed for device discovery?

AlanGriffiths, 2021-12-06:
@EinarArnason sorry, nobody around here is experienced with using Virtualbox.

There is no reason why Mir wouldn't work with Virtualbox provided there's a working gbm/kms graphics stack. I've seen it working many times in the past. I know VM graphics can be affected by the bios, host system, Virtualbox configuration, guest system and Virtualbox "guest modules". But I don't have a working configuration to compare with.

Saviq, 2022-10-21:
I recently ran Mir and Frame just fine on VirtualBox, provided that 3D acceleration was enabled.

closed 2022-10-21 by Saviq