canonical/ubuntu-frame #48: Popups are not dismissed after re-clicking the button that triggers them

wmww, 2021-12-17:
It appears https://github.com/MirServer/mir/issues/1818 is still present or has reemerged on frame. The frame window manager has this bug even outside of the snap. miral-shell, using the exact same Mir, does not have the bug.

AlanGriffiths, 2022-01-04:
This is even weirder than the referenced bug...

    export WAYLAND_DISPLAY=wayland-99
    ubuntu-frame&
    xfce4-terminal

**Expect:**
Clicking on the "Edit" menu brings up the "Edit" menu.
Clicking again causes the menu to close.

**Actual:**
Clicking on the "Edit" menu brings up the "File" menu.
Clicking again causes the menu to "flash".

AlanGriffiths, 2022-01-04:
Ah! https://github.com/MirServer/mir/pull/1948 fixed this in `MinimalWindowManager`, not in `CanonicalWindowManagerPolicy`.

AlanGriffiths, 2022-01-04:
Regardless, this behaviour shouldn't be a "policy" option, it ought to be built-in (e.g. in `miral::BasicWindowManager`). Should we "fix" Ubuntu Frame? Or Mir?

AlanGriffiths, 2022-01-05:
OK, the "wrong menu" opening problem isn't Frame related. Raised here: https://github.com/MirServer/mir/issues/2268

commit c0bcace0 (2022-01-10): Use MinimalWindowManager as base (#51)

Using MinimalWindowManager as the basis of the FrameWindowManagerPolicy means we need to implement less functionality and support for non top-level windows (e.g. dialogs) being moved and resized.

Fixes: #48