canonical/ubuntu-frame #49: Flutter app can't create a GL context on RPi 4/Ubuntu Core 20

KirioXX, 2022-01-03:
Hi,
we are in the process to port our flutter app to Ubuntu Core with Ubuntu Frame.
But we sadly see an error when we try to start our app and the screen stays white.
What prevents us from shipping our devices with Ubuntu Core.

These are the logs that we see:
```
2021-11-26T11:57:34Z trunk-pilot-dashboard.daemon[1071]: + snapctl get daemon
2021-11-26T11:57:34Z trunk-pilot-dashboard.daemon[773]: + [ true = true ]
2021-11-26T11:57:34Z trunk-pilot-dashboard.daemon[773]: + exec /snap/trunk-pilot-dashboard/x1/bin/wayland-launch /snap/trunk-pilot-dashboard/x1/bin/trunk_pilot_dashboard
2021-11-26T11:57:35Z trunk-pilot-dashboard.daemon[1083]: error: error running snapctl: snap "trunk-pilot-dashboard" has no plug or slot named "pulseaudio"
2021-11-26T11:57:35Z trunk-pilot-dashboard.daemon[773]: WARNING: pulseaudio interface not connected! Please run: /snap/trunk-pilot-dashboard/current/bin/setup.sh
2021-11-26T11:57:35Z trunk-pilot-dashboard.daemon[1098]: Setting up watches.
2021-11-26T11:57:35Z trunk-pilot-dashboard.daemon[1098]: Watches established.
2022-01-03T10:03:44Z trunk-pilot-dashboard.daemon[1098]: /run/user/0/ CREATE wayland-0.lock
2022-01-03T10:03:45Z -[773]: g_dbus_proxy_new_sync: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
2022-01-03T10:03:49Z -[773]: Failed to start Flutter renderer: Unable to create a GL context
```

We followed the ["RUNNING_ON_YOUR_DEVICE" documentation](https://github.com/MirServer/ubuntu-frame/blob/master/RUNNING_ON_YOUR_DEVICE.md) to set up Ubuntu Frame and ["Run Flutter Applications on Ubuntu Core"](https://ubuntu.com/tutorials/run-flutter-applications-on-ubuntu-core) to configure our flutter app.

Here is also our snap config:
```
name: trunk-andon
version: 1.0.3
summary: Trunk Andon App
description: Trunk Andon app to visualize pilot data.

confinement: strict
base: core18
grade: stable

apps:
  daemon:
    daemon: simple
    restart-condition: always
    command-chain:
      - bin/run-daemon
      - bin/wayland-launch
    command: bin/trunk-andon
    extensions: [flutter-master]

  trunk-andon:
    command-chain:
      - bin/wayland-launch
    command: trunk-andon
    extensions: [flutter-master]

parts:
  trunk-andon:
    source: .
    plugin: flutter
    flutter-target: lib/main.dart

  mir-kiosk-snap-launch:
    plugin: dump
    source: https://github.com/MirServer/mir-kiosk-snap-launch.git
    override-build: $SNAPCRAFT_PART_BUILD/build-with-plugs.sh opengl pulseaudio wayland
    stage-packages:
      - inotify-tools

  assets:
    plugin: nil
    stage-packages:
      - dmz-cursor-theme
      - fonts-dejavu
      - fonts-freefont-ttf
      - fonts-ubuntu

layout:
  /usr/share/fonts:
    bind: $SNAP/usr/share/fonts
  /etc/fonts:
    bind: $SNAP/etc/fonts
  /usr/share/icons:
    bind: $SNAP/usr/share/icons
```
Did we configure something wrong or do we need to connect the app with wayland manually to make it work?
Thank you!

AlanGriffiths, 2022-01-03:
@KirioXX, I'm not sure why your app isn't starting but the logs show that Wayland is connected. You can confirm this with `snap connections trunk-andon`.

There is a unrelated warning:
```
2021-11-26T11:57:35Z trunk-pilot-dashboard.daemon[773]: WARNING: pulseaudio interface not connected! Please run: /snap/trunk-pilot-dashboard/current/bin/setup.sh
```
This is because you list `pulseaudio` in your `mir-kiosk-snap-launch` stanza, but don't use a pulseaudio plug.

The following guide doesn't yet cover Flutter (you followed the right tutorial), but the process shown may help track down your issue:

https://github.com/MirServer/iot-example-graphical-snap/wiki

KirioXX, 2022-01-03:
@AlanGriffiths thank you for the quick response.
It looks like the app is connected to wayland:
```sh
$ snap connections trunk-pilot-dashboard
Interface                 Plug                                   Slot                             Notes
content[gnome-3-28-1804]  trunk-pilot-dashboard:gnome-3-28-1804  gnome-3-28-1804:gnome-3-28-1804  -
content[gtk-3-themes]     trunk-pilot-dashboard:gtk-3-themes     gtk-common-themes:gtk-3-themes   -
content[icon-themes]      trunk-pilot-dashboard:icon-themes      gtk-common-themes:icon-themes    -
content[sound-themes]     trunk-pilot-dashboard:sound-themes     gtk-common-themes:sound-themes   -
desktop                   trunk-pilot-dashboard:desktop          -                                -
desktop-legacy            trunk-pilot-dashboard:desktop-legacy   -                                -
gsettings                 trunk-pilot-dashboard:gsettings        -                                -
opengl                    trunk-pilot-dashboard:opengl           :opengl                          -
wayland                   trunk-pilot-dashboard:wayland          ubuntu-frame:wayland             -
x11                       trunk-pilot-dashboard:x11              -
```
I'm not sure if it could be related but I follow this issue for a while now https://github.com/flutter/flutter/issues/76178 because I had a similar issue on ubuntu desktop a while back. 

The pulsaudio is not an issue we just need a picture and network for now, but I left it in to look into it in case we decide we need audio in the future.

bugraaydogar, 2022-01-04:
Hi @KirioXX,

Did you also run into the similar issue within qemu or x86 based system? I'm afraid it could be either platform specific or the application specific problem?

Are you able to run basic hello-world application without any issue?

AlanGriffiths, 2022-01-04:
> I'm not sure if it could be related but I follow this issue for a while now [flutter/flutter#76178](https://github.com/flutter/flutter/issues/76178) because I had a similar issue on ubuntu desktop a while back.

That looks unrelated, the Mesa drivers (including swrast) will be in `$SNAP/gnome-platform/usr/lib/$SNAP_LAUNCHER_ARCH_TRIPLET/dri`

KirioXX, 2022-01-04:
Hi @bugraaydogar,

I have tried it on qemu with a Ubuntu Core arm64 VM sadly with the same result, I couldn't test it on x86 because I was running in this https://github.com/MirServer/ubuntu-frame/issues/31 issue.

I also just tried the example app in the ["Run Flutter Applications on Ubuntu Core"](https://ubuntu.com/tutorials/run-flutter-applications-on-ubuntu-core#1-overview) with the same result:
```sh
2022-01-04T10:36:02Z systemd[1]: Started Service for snap application super-cool-app.daemon.
2022-01-04T10:36:02Z super-cool-app.daemon[2199]: + snapctl get daemon
2022-01-04T10:36:02Z super-cool-app.daemon[2082]: + [ true = true ]
2022-01-04T10:36:02Z super-cool-app.daemon[2082]: + exec /snap/super-cool-app/x1/bin/wayland-launch /snap/super-cool-app/x1/bin/flutterdemo
2022-01-04T10:36:03Z -[2082]: g_dbus_proxy_new_sync: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
2022-01-04T10:36:20Z -[2082]: Failed to start Flutter renderer: Unable to create a GL context
```

bugraaydogar, 2022-01-04:
@KirioXX what about running on qemu with amd64 rather than arm64 ? 

I'm personally able to run the example application without any issue. My setup is;
* Ubuntu Desktop 20.04 for building the snap.
* Ubuntu Core 20 with qemu on my Ubuntu Desktop 20.04 (https://ubuntu.com/download/kvm)

I still believe that there is an issue with ARM based platforms.

AlanGriffiths, 2022-01-04:
> I have tried it on qemu with a Ubuntu Core arm64 VM sadly with the same result, I couldn't test it on x86 because I was running in this #31 issue.

Thanks for reminding me about that issue, there's a workaround (which I've added).

> I also just tried the example app in the ["Run Flutter Applications on Ubuntu Core"](https://ubuntu.com/tutorials/run-flutter-applications-on-ubuntu-core#1-overview) with the same result:

That suggests there may be a problem with Flutter apps on RPi4. I know they are based on core18 - maybe that lacks suitable drivers. I don't have a RPi4 to test, will see if I can find someone that can.

----

Note: I'm going to close this issue as it clearly isn't a problem with ubuntu-frame itself. (You can still post updates here and we'll see them.)

closed 2022-01-04 by AlanGriffiths

AlanGriffiths, 2022-01-04:
@RAOF I believe you've got a RPi4: could you check the ["Run Flutter Applications on Ubuntu Core"](https://ubuntu.com/tutorials/run-flutter-applications-on-ubuntu-core#1-overview) works with that kit?

AlanGriffiths, 2022-01-07:
@KirioXX I've got hold of an RPi400 and can reproduce the problem you see. I also have a workaround.

The Flutter demo starts, but reports (as you see)
```
... Failed to start Flutter renderer: Unable to create a GL context
```
But it does display a white screen. (So it obviously managed to render that much before hitting problems.)

My workaround is to force Mesa to do software rendering by adding `LIBGL_ALWAYS_SOFTWARE: 1` to `environment`. That gets things rendering correctly (but uses software rendering).

This suggests that there's a problem with the `v3d_dri` Mesa driver bundled with `gnome-3-28-1804` when used on Core20/RPi4. I'll investigate further, but hopefully the workaround can unblock you while I investigate the underlying issue.

AlanGriffiths, 2022-01-07:
[Update on investigation]

I've not yet proved it, but it looks like the problem originates with the `base: core18` required by the `flutter-master` extension. This pulls in an older version of Mesa (comparing against a snap with the `core20` base):

```
$ grep libgl1-mesa-dri  /snap/gnome-3-28-1804/current/snap/manifest.yaml 
    - libgl1-mesa-dri=20.0.8-0ubuntu1~18.04.1
- libgl1-mesa-dri=20.0.8-0ubuntu1~18.04.1
$ grep libgl1-mesa-dri  /snap/mesa-core20/current/snap/manifest.yaml 
    override-pull: snapcraftctl set-version `LANG=C apt-cache policy libgl1-mesa-dri
    - libgl1-mesa-dri=21.0.3-0ubuntu0.2~20.04.1
- libgl1-mesa-dri=21.0.3-0ubuntu0.2~20.04.1
```
Both versions have the `v3d_dri` that _should_ work:
```
$ ls -lh /snap/mesa-core20/current/egl/dri/v3d_dri.so /snap/gnome-3-28-1804/current/usr/lib/aarch64-linux-gnu/dri/v3d_dri.so
-rw-r--r-- 33 root root 16M Jun 12  2020 /snap/gnome-3-28-1804/current/usr/lib/aarch64-linux-gnu/dri/v3d_dri.so
-rw-r--r-- 37 root root 20M May 25  2021 /snap/mesa-core20/current/egl/dri/v3d_dri.so
```
As there's a significant size difference, there must have been some changes, and these may well explain the difference in behaviour.

KirioXX, 2022-01-07:
@AlanGriffiths that's amazing, just tried it on my test pi. Thank you very much 🙌

I'm not sure how far the core 20 support for flutter is.
It looks like that there was no movement for a month on the PR https://github.com/canonical/flutter-snap/pull/61

AlanGriffiths, 2022-01-07:
[Update 2 on investigation]

I've built a [core20 based Flutter demo snap](https://github.com/MirServer/iot-example-graphical-snap/tree/Flutter-demo) that uses Mesa 21. But this doesn't fix the problem.

Conclusion: it isn't the Mesa version from 18.04 that's the problem. That theory was wrong.

AlanGriffiths, 2022-01-07:
[Update 3 on investigation]

Running on amd64/intel:
```
$ GDK_DEBUG=opengl snap run iot-example-graphical-snap

(flutterdemo:120983): Gtk-WARNING **: 17:12:58.070: Locale not supported by C library.
        Using the fallback 'C' locale.
Gdk-Message: 17:12:58.190: EGL API version 1.5 found
 - Vendor: Mesa Project
 - Version: 1.5
 - Client APIs: OpenGL OpenGL_ES 
 - Extensions:
        EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_create_context_robustness EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_EXT_swap_buffers_with_damage EGL_IMG_context_priority EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image_base EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_KHR_swap_buffers_with_damage EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display EGL_WL_create_wayland_buffer_from_image 
Gdk-Message: 17:12:58.190: Creating EGL context version 3.2 (debug:no, forward:no, legacy:no, es:no)
...
```

Running on arm64/ARM Cortex-A72:
```
$ sudo GDK_DEBUG=opengl snap run iot-example-graphical-snap
Gdk-Message: 17:16:47.531: EGL API version 1.4 found
 - Vendor: Mesa Project
 - Version: 1.4
 - Client APIs: OpenGL OpenGL_ES 
 - Extensions:
	EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_EXT_swap_buffers_with_damage EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image_base EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_KHR_swap_buffers_with_damage EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display EGL_WL_create_wayland_buffer_from_image 
Gdk-Message: 17:16:47.531: Creating EGL context version 3.2 (debug:no, forward:no, legacy:no, es:no)
Gdk-Message: 17:16:47.532: eglCreateContext failed, switching to legacy

** (flutterdemo:8828): WARNING **: 17:16:47.532: Failed to start Flutter renderer: Unable to create a GL context
```

So, we're encountering an earlier EGL API, with fewer extension. (Specifically missing EGL_ANDROID_native_fence_sync, EGL_EXT_create_context_robustness, EGL_IMG_context_priority - none of which are obviously an issue)

AlanGriffiths, 2022-01-07:
@KirioXX the best answer seems to be to add `GDK_GL: gles` it seems that the v3d driver doesn't support legacy GL contexts. But it does support gles.

KirioXX, 2022-01-10:
@AlanGriffiths Perfect I tried it over the weekend and it works like a charm.