canonical/ubuntu-frame #55: Compatibility issues with Intel i915

mtinnes, 2022-01-31:
Having issues enabling Frame and wondering where in the software stack the issue lies. When switching to daemon=true the screen switches to console mode with the following msg:

`[  132.546752 ] i915 [drm] *ERROR* failed to enable link training`

Our hardware is Intel i3 with Intel UHD Graphics 620 running Ubuntu 21.10 desktop version.

I'm guessing this is a video driver issue, but would like to know where to start looking for a fix.

AlanGriffiths, 2022-01-31:
Please attach the logs from ubuntu-frame:
```
snap logs -n 100 ubuntu-frame
```

mtinnes, 2022-01-31:
Thanks for the quick reply @AlanGriffiths. We did get a signal (grey gradient background) after connecting to the display port. We have 3 video outs (2 HDMI, 1 DP). No doubt there is a configuration setting for making the 1st HDMI port the primary output for Frame?

Saviq, 2022-01-31:
Hi @mtinnes, have a look here for how to configure displays:

https://github.com/MirServer/ubuntu-frame/blob/master/REFERENCE.md#the-display-snap-configuration-option

Did you run Frame prior to connecting the displays as desired? If so, check the contents of the `frame.display` file for suggestions. If Mir doesn't see a display connected over HDMI, that's more likely a hardware issue, not a software one.

If you can provide logs from when your display is connected over HDMI, we could say more.

mtinnes, 2022-01-31:
It's odd, the following log is from starting Frame with HDMI attached. The desktop is visible at the full 1920x1200 resolution over HDMI, however looking at the logs is appears the only devices connected are **eDP-1** and **DisplayPort-2**.

```
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.912600] <information> evdev-input: Added /dev/input/event2
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.915455] <information> evdev-input: Added /dev/input/event3
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.918107] <information> evdev-input: Added /dev/input/event1
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.920635] <information> evdev-input: Added /dev/input/event0
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.926266] <information> evdev-input: Added /dev/input/event4
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.930246] <information> evdev-input: Added /dev/input/event9: USB Receiver
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.930684] < -warning- > gbm-kms: move_cursor: drmModeMoveCursor failed (Permission denied)
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.930716] < -warning- > gbm-kms: set_cursor: drmModeSetCursor failed (Permission denied)
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.930746] <information> evdev-input: Added /dev/input/event5: Wireless Device
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.933258] <information> evdev-input: Added /dev/input/event16: Wireless Device
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.935740] <information> evdev-input: Added /dev/input/event6: Wireless Device
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.938189] <information> evdev-input: Added /dev/input/event7: Wireless Device
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.940721] <information> evdev-input: Added /dev/input/event8: Wireless Device
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.940799] <information> evdev-input: Added /dev/input/event10: USB Receiver
2022-01-31T16:24:02-06:00 ubuntu-frame.daemon[17716]: [2022-01-31 16:24:02.943527] < -warning- > evdev-input: Attempted to remove unmanaged fd 53
2022-01-31T16:24:03-06:00 systemd[1]: snap.ubuntu-frame.daemon.service: Main process exited, code=killed, status=6/ABRT
2022-01-31T16:24:03-06:00 systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'signal'.
2022-01-31T16:24:06-06:00 systemd[1]: snap.ubuntu-frame.daemon.service: Scheduled restart job, restart counter is at 4.
2022-01-31T16:24:06-06:00 systemd[1]: Stopped Service for snap application ubuntu-frame.daemon.
2022-01-31T16:24:06-06:00 systemd[1]: Started Service for snap application ubuntu-frame.daemon.
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: + exec nohup /snap/ubuntu-frame/746/bin/run-frame /snap/ubuntu-frame/746/usr/local/bin/frame
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: + [ /snap/ubuntu-frame/746/usr/local/bin/frame = --help ]
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17792]: + dirname /run/user/0/snap.ubuntu-frame
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: + XDG_RUNTIME_DIR=/run/user/0
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: + export XDG_RUNTIME_DIR
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: + mkdir -p /run/user/0 -m 700
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: + [ -n  ]
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: + exec /snap/ubuntu-frame/746/usr/local/bin/frame
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.698015] <information> mirserver: Starting
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.977676] < - debug - > mirserver: Using Linux VT subsystem for session management
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.978271] <information> mircommon: Loading modules from: /snap/ubuntu-frame/746/usr/lib/x86_64-linux-gnu/mir/server-platform
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.978490] <information> mircommon: Loading module: /snap/ubuntu-frame/746/usr/lib/x86_64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.19
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.978585] <information> mircommon: Loading module: /snap/ubuntu-frame/746/usr/lib/x86_64-linux-gnu/mir/server-platform/server-x11.so.19
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.978655] <information> mircommon: Loading module: /snap/ubuntu-frame/746/usr/lib/x86_64-linux-gnu/mir/server-platform/input-evdev.so.8
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.991268] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card0, driver i915
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.992254] <information> mirserver: Found graphics driver: mir:gbm-kms (version 2.6.0) Support priority: 256
2022-01-31T16:24:06-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:06.992431] <information> mirserver: Found graphics driver: mir:x11 (version 2.6.0) Support priority: 0
[... cut ...]
2022-01-31T16:24:07-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:07.433025] <information> evdev-input: Added /dev/input/event16: Wireless Device
2022-01-31T16:24:07-06:00 ubuntu-frame.daemon[17774]: [2022-01-31 16:24:07.435923] < -warning- > evdev-input: Attempted to remove unmanaged fd 49
2022-01-31T16:24:07-06:00 systemd[1]: snap.ubuntu-frame.daemon.service: Main process exited, code=killed, status=6/ABRT
2022-01-31T16:24:07-06:00 systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'signal'.
2022-01-31T16:24:09-06:00 systemd[1]: Stopped Service for snap application ubuntu-frame.daemon.

```

Saviq, 2022-02-01:
@mtinnes so does everything work as expected, other than the output labels? DisplayPort is trivially convertible to HDMI, so maybe the hardware misreports which output is what.