canonical/ubuntu-frame #62: Cannot start wpe-webkit-mir-kiosk on Ubuntu Server 22.04

lejsekt, 2022-05-10:
Hi,
I'm trying to follow the startup guide on Ubuntu Server 22.04 without success. This is what I get after issuing `sudo snap start wpe-webkit-mir-kiosk`:

```
2022-05-10T07:24:08Z -[8910]: Could not open /sys/class/dmi/id/chassis_type: Failed to open file “/sys/class/dmi/id/chassis_type”: Permission denied
2022-05-10T07:24:08Z -[8910]: Could not open /sys/firmware/acpi/pm_profile: Failed to open file “/sys/firmware/acpi/pm_profile”: Permission denied
2022-05-10T07:24:08Z -[8910]: Acquired D-Bus well-known name com.igalia.Cog
2022-05-10T07:24:09Z wpe-webkit-mir-kiosk.daemon[8832]: /snap/wpe-webkit-mir-kiosk/63/bin/launch-wpe: line 27:  8910 Segmentation fault      (core dumped) "$SNAP"/usr/bin/cog -P fdo --bg-color=black --enable-mediasource=1 --webprocess-failure=restart --enable-write-console-messages-to-stdout="$error_to_console" "$url"
2022-05-10T07:24:09Z systemd[1]: snap.wpe-webkit-mir-kiosk.daemon.service: Main process exited, code=exited, status=139/n/a
2022-05-10T07:24:09Z systemd[1]: snap.wpe-webkit-mir-kiosk.daemon.service: Failed with result 'exit-code'.
```

Any ideas?

I've installed ubuntu-frame and it started succesfully.

Versions:
ubuntu-frame: 61-mir2.7.0
wpe-webkit-mir-kiosk: 2.30.5

Thank you

AlanGriffiths, 2022-05-10:
Hi @lejsekt, this is issues list for `ubuntu-frame` which you say started successfully. We don't maintain `wpe-webkit-mir-kiosk`, it is [a community project](https://gitlab.com/glancr/wpe-webkit-snap/). Because of this I don't know if the `Permission denied` messages are significant or not.

We've not encountered issues with `wpe-webkit-mir-kiosk` but as you've not stated which processor architecture or device you are running on I don't know if we've ever tried the same setup. (Also, we don't generally test external projects on Ubuntu Server.) 

If the `Permission denied` messages are significant you might get a clue from `dmesg` or `snappy-debug.scanlog`, but the best place to ask is on the [wpe-webkit-snap issues list](https://gitlab.com/glancr/wpe-webkit-snap/-/issues) or, possibly, the [Snapcraft forum](https://forum.snapcraft.io/search?q=wpe-webkit-mir-kiosk%20category%3A14)

closed 2022-05-10 by AlanGriffiths

Saviq, 2022-05-10:
For some anecdata, I was just able to run Frame with WPE on 22.04 desktop as well as server.