canonical/ubuntu-frame #65: ubuntu-frame - after a few hours running the services are stopped automatically

bosgold, 2022-05-28:
ubuntu-frame with wpe-webkit-mir-kiosk works well after a reboot.  But after a few hours service are automatically shut down. And the screen becomes grey.  

Ubuntu 22.04 LTS (GNU/Linux 5.15.0-33-generic x86_64)

# snap list
```
Name                  Version                 Rev    Tracking       Publisher   Notes
core20                20220512                1494   latest/stable  canonical✓  base
lxd                   5.0.0-b0287c1           22923  5.0/stable/…   canonical✓  -
mesa-core20           21.0.3                  95     latest/stable  canonical✓  -
network-manager       1.22.10-13              735    20/stable      canonical✓  -
snapcraft             6.1                     7201   latest/stable  canonical✓  classic
snapd                 2.55.5                  15904  latest/stable  canonical✓  snapd
ubuntu-frame          61-mir2.7.0             1902   latest/stable  canonical✓  -
ubuntu-frame-osk      38-squeekboard-v1.17.1  115    latest/stable  canonical✓  -
wpe-webkit-mir-kiosk  2.30.5                  63     latest/stable  glancr      -
```

/var/log/syslog
May 28 20:42:48 ausstellung systemd[1]: session-14.scope: Deactivated successfully.
May 28 20:42:58 ausstellung systemd[1]: Stopping User Manager for UID 0...
May 28 20:42:58 ausstellung systemd[2234]: Stopped target Main User Target.
May 28 20:42:58 ausstellung systemd[2234]: Stopped target Basic System.
May 28 20:42:58 ausstellung systemd[2234]: Stopped target Paths.
May 28 20:42:58 ausstellung systemd[2234]: Stopped target Sockets.
May 28 20:42:58 ausstellung systemd[2234]: Stopped target Timers.
May 28 20:42:58 ausstellung systemd[2234]: Closed D-Bus User Message Bus Socket.
May 28 20:42:58 ausstellung systemd[2234]: Closed GnuPG network certificate management daemon.
May 28 20:42:58 ausstellung systemd[2234]: Closed GnuPG cryptographic agent and passphrase cache (access for web browsers).
May 28 20:42:58 ausstellung systemd[2234]: Closed GnuPG cryptographic agent and passphrase cache (restricted).
May 28 20:42:58 ausstellung systemd[2234]: Closed GnuPG cryptographic agent (ssh-agent emulation).
May 28 20:42:58 ausstellung systemd[2234]: Closed GnuPG cryptographic agent and passphrase cache.
May 28 20:42:58 ausstellung systemd[2234]: Closed debconf communication socket.
May 28 20:42:58 ausstellung systemd[2234]: Closed REST API socket for snapd user session agent.
May 28 20:42:58 ausstellung systemd[2234]: Removed slice User Application Slice.
May 28 20:42:58 ausstellung systemd[2234]: Reached target Shutdown.
May 28 20:42:58 ausstellung systemd[2234]: Finished Exit the Session.
May 28 20:42:58 ausstellung systemd[2234]: Reached target Exit the Session.
May 28 20:42:58 ausstellung systemd[1]: user@0.service: Deactivated successfully.
May 28 20:42:58 ausstellung systemd[1]: Stopped User Manager for UID 0.
May 28 20:42:58 ausstellung systemd[1]: Stopping User Runtime Directory /run/user/0...
May 28 20:42:58 ausstellung systemd[1]: user-runtime-dir@0.service: Deactivated successfully.
May 28 20:42:58 ausstellung systemd[1]: Stopped User Runtime Directory /run/user/0.
May 28 20:42:58 ausstellung systemd[1]: Removed slice User Slice of UID 0.
May 28 20:42:58 ausstellung systemd[1]: user-0.slice: Consumed 1.233s CPU time.
May 28 20:42:58 ausstellung systemd[1]: snap.wpe-webkit-mir-kiosk.restart-watcher.service: Deactivated successfully.
May 28 20:42:58 ausstellung systemd[1]: snap.wpe-webkit-mir-kiosk.restart-watcher.service: Scheduled restart job, restart counter is at 1.
May 28 20:42:58 ausstellung systemd[1]: Stopped Service for snap application wpe-webkit-mir-kiosk.restart-watcher.
May 28 20:42:58 ausstellung systemd[1]: Started Service for snap application wpe-webkit-mir-kiosk.restart-watcher.
May 28 20:42:58 ausstellung wpe-webkit-mir-kiosk.restart-watcher[4979]: Wayland socket not available, waiting one second

Restarting 
# snap restart ubuntu-frame
fixes the problem.  The wpe-webkit-mir-kiosk.restart-watcher restarts the kiosk. For another few hours until the same happens again.

Saviq, 2022-05-29:
Hi @bosgold, all the lines about "stopped" / "closed" say that it's something external to Ubuntu Frame (through systemd) is stopping it. Like there's a timeout set up somewhere.

Showing more logs above what's there may help. Also maybe looking at `sudo systemctl status`.

I'll run this setup myself here today to see if I can reproduce in any way.

bosgold, 2022-05-29:
> /var/log/syslog

`May 29 14:09:07 ausstellung systemd[1]: Starting Clean php session files...
May 29 14:09:07 ausstellung systemd[1]: phpsessionclean.service: Deactivated successfully.
May 29 14:09:07 ausstellung systemd[1]: Finished Clean php session files.
May 29 14:11:31 ausstellung systemd[1]: session-2.scope: Deactivated successfully.
May 29 14:11:41 ausstellung systemd[1]: Stopping User Manager for UID 0...
May 29 14:11:41 ausstellung systemd[2126]: Stopped target Main User Target.
May 29 14:11:41 ausstellung systemd[2126]: Stopped target Basic System.
May 29 14:11:41 ausstellung systemd[2126]: Stopped target Paths.
May 29 14:11:41 ausstellung systemd[2126]: Stopped target Sockets.
May 29 14:11:41 ausstellung systemd[2126]: Stopped target Timers.
May 29 14:11:41 ausstellung systemd[2126]: Closed D-Bus User Message Bus Socket.
May 29 14:11:41 ausstellung systemd[2126]: Closed GnuPG network certificate management daemon.
May 29 14:11:41 ausstellung systemd[2126]: Closed GnuPG cryptographic agent and passphrase cache (access for web browsers).
May 29 14:11:41 ausstellung systemd[2126]: Closed GnuPG cryptographic agent and passphrase cache (restricted).
May 29 14:11:41 ausstellung systemd[2126]: Closed GnuPG cryptographic agent (ssh-agent emulation).
May 29 14:11:41 ausstellung systemd[2126]: Closed GnuPG cryptographic agent and passphrase cache.
May 29 14:11:41 ausstellung systemd[2126]: Closed debconf communication socket.
May 29 14:11:41 ausstellung systemd[2126]: Closed REST API socket for snapd user session agent.
May 29 14:11:41 ausstellung systemd[2126]: Removed slice User Application Slice.
May 29 14:11:41 ausstellung systemd[2126]: Reached target Shutdown.
May 29 14:11:41 ausstellung systemd[2126]: Finished Exit the Session.
May 29 14:11:41 ausstellung systemd[2126]: Reached target Exit the Session.
May 29 14:11:41 ausstellung systemd[1]: user@0.service: Deactivated successfully.
May 29 14:11:41 ausstellung systemd[1]: Stopped User Manager for UID 0.
May 29 14:11:41 ausstellung systemd[1]: Stopping User Runtime Directory /run/user/0...
May 29 14:11:41 ausstellung systemd[1]: user-runtime-dir@0.service: Deactivated successfully.
May 29 14:11:41 ausstellung systemd[1]: Stopped User Runtime Directory /run/user/0.
May 29 14:11:41 ausstellung systemd[1]: Removed slice User Slice of UID 0.
May 29 14:11:41 ausstellung systemd[1]: snap.wpe-webkit-mir-kiosk.restart-watcher.service: Deactivated successfully.
May 29 14:11:41 ausstellung systemd[1]: snap.wpe-webkit-mir-kiosk.restart-watcher.service: Consumed 1.210s CPU time.
May 29 14:11:41 ausstellung systemd[1]: snap.wpe-webkit-mir-kiosk.restart-watcher.service: Scheduled restart job, restart counter is at 1.
May 29 14:11:41 ausstellung systemd[1]: Stopped Service for snap application wpe-webkit-mir-kiosk.restart-watcher.
May 29 14:11:41 ausstellung systemd[1]: snap.wpe-webkit-mir-kiosk.restart-watcher.service: Consumed 1.210s CPU time.
May 29 14:11:41 ausstellung systemd[1]: Started Service for snap application wpe-webkit-mir-kiosk.restart-watcher.
May 29 14:11:41 ausstellung kernel: [ 1389.585780] kauditd_printk_skb: 14 callbacks suppressed
May 29 14:11:41 ausstellung kernel: [ 1389.585783] audit: type=1400 audit(1653826301.610:119): apparmor="DENIED" operation="capable"
 profile="/snap/snapd/15904/usr/lib/snapd/snap-confine" pid=2406 comm="snap-confine" capability=12  capname="net_admin"
May 29 14:11:41 ausstellung kernel: [ 1389.585786] audit: type=1400 audit(1653826301.610:120): apparmor="DENIED" operation="capable"
 profile="/snap/snapd/15904/usr/lib/snapd/snap-confine" pid=2406 comm="snap-confine" capability=38  capname="perfmon"
May 29 14:11:41 ausstellung wpe-webkit-mir-kiosk.restart-watcher[2406]: Wayland socket not available, waiting one second
May 29 14:12:00 ausstellung wpe-webkit-mir-kiosk.restart-watcher[2406]: message repeated 19 times: [ Wayland socket not available, waiting one second]
`

> "systemctl status" before the loss of the Wayland socket
`ausstellung
    State: running
     Jobs: 0 queued
   Failed: 0 units
    Since: Sun 2022-05-29 13:48:36 CEST; 58s ago
   CGroup: /
           +-user.slice
           ¦ +-user-0.slice
           ¦   +-session-2.scope
           ¦   ¦ +-2123 sshd: root@pts/0
           ¦   ¦ +-2211 -bash
           ¦   ¦ +-2222 systemctl status
           ¦   ¦ +-2223 pager
           ¦   +-user@0.service …
           ¦     +-init.scope
           ¦       +-2126 /lib/systemd/systemd --user
           ¦       +-2127 (sd-pam)
           +-init.scope
           ¦ +-1 /sbin/init
           +-system.slice
             +-irqbalance.service
             ¦ +-1273 /usr/sbin/irqbalance --foreground
             +-apache2.service
             ¦ +-1402 /usr/sbin/apache2 -k start
[... 139 lines cut ...]
             +-dbus.service
             ¦ +-1268 @dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
             ¦ +-2007 /lib/netplan/netplan-dbus
             +-systemd-timesyncd.service
             ¦ +-1175 /lib/systemd/systemd-timesyncd
             +-snap.wpe-webkit-mir-kiosk.restart-watcher.service
             ¦ +- 2406 /bin/sh -e /snap/wpe-webkit-mir-kiosk/63/bin/watcher
             ¦ +-13219 sleep 1
             +-system-getty.slice
             ¦ +-getty@tty1.service
             ¦   +-1408 /sbin/agetty -o -p -- \u --noclear tty1 linux
             +-snap.wpe-webkit-mir-kiosk.daemon.service
[... cut ...]
The problem seems to be triggered when systemd cleans up the SSH session (slice) for user root I opened to check the system. Within this process (PID 2126 /lib/systemd/systemd --user) is stopped this step
May 29 14:11:41 ausstellung systemd[1]: Stopped User Runtime Directory /run/user/0.
does obiously also remove the Wayland socket located in this directory. 
Does by the way not directly result directly in a grey screen. This comes only later when wpe-webkit-mir-kiosk is triggered to reconnect to Wayland (snap restart wpe-webkit-mir-kiosk).

Saviq, 2022-05-29:
@bosgold thanks, I can reproduce, the workaround is to _not_ ssh in as root, but rather as a user and use `sudo`.

We'll try and work out who's at fault, regardless.

Saviq, 2022-05-30:
This needs to be solved on the snapd side, I wrote something up on an existing forum discussion:

https://forum.snapcraft.io/t/rethinking-how-we-handle-xdg-runtime-dir/22223/14

Will close this as it's not something we can solve on Frame's side.

closed 2022-05-30 by Saviq