canonical/ubuntu-frame #66: How to EXIT the locked screen?

SomeDevWeb, 2022-06-01:
Sorry, it might be a silly question, but I didn't find any info about this. 
So I installed Ubuntu 22 Desktop (minimal config) and I successfully installed ubuntu-frame and mir server. 
When I ran "**snap set ubuntu-frame daemon=true**" the screen locked and I have no idea how to get out of it. 
I restarted it and, as it should do, it comes back to the locked screen, but I didn;t even get to change the URL or anything else, it's just the grey screen. 
Usually, the kiosk locking apps have a key combination or a touch sequence to show an admin password. 
How does ubuntu-frame work in that respect?
Thank you so much!

Saviq, 2022-06-01:
Hi @SomeDevWeb, you can use Ctrl+Alt+F* to switch to another virtual terminal and log in there. It's generally preferred to use remote access through SSH for management, though.

We will add better documentation about this. We also may need to think about optionally _locking_ to the Frame virtual terminal - for kiosks that have a keyboard.

SomeDevWeb, 2022-06-01:
Hi. Thanks for the quick reply! Really appreciate it. 
Indeed, I pressed ctrl+alt+f3 and I got to another ubuntu screen, then I added the URL setting. 
But after that I restarted and there was nothing shown. Still the grey screen...
Should I set something else up? 
BTW, what browser does it use?
Thanks again!

Saviq, 2022-06-01:
@SomeDevWeb I'm assuming you're following something like this tutorial?

https://ubuntu.com/tutorials/secure-ubuntu-kiosk

If so, it uses a dedicated embedded web view based on the WebKit engine:

https://snapcraft.io/wpe-webkit-mir-kiosk

As for why it doesn't display, did you run `sudo /snap/wpe-webkit-mir-kiosk/bin/setup.sh`? On Ubuntu Desktop, that you say you installed, there's a bit more setup necessary (we recommend Ubuntu Core).

You can see helpful logs in `journalctl`.

AlanGriffiths, 2022-06-01:
@SomeDevWeb the discussion of dwservice.net is off topic for "How to EXIT the locked screen?". It is probably better suited to the [Mir forum](https://discourse.ubuntu.com/c/mir/15) than an unrelated Frame "issue".

I'm also unclear what you are trying to do with it. Even if the technical difficulties were solved, accessing the shell that Frame provides via a dwservice.net agent would give very limited access to the system as Frame doesn't provide a "Desktop Enviroment".

Closing issue as the original problem appears to be resolved.

closed 2022-06-01 by AlanGriffiths

SomeDevWeb, 2022-06-01:
Yes, the original issue is solved, thanks. 
As a thought, I also found a way to disable Wayland which gives a nice option at login to log as Xorg and I can fully do what I need with the remote work.