canonical/ubuntu-frame #68: Put display to sleep

ghanjiboy, 2022-06-11:
Hi, I am using ubuntu frame with mir kiosk - works great.  I would like to put the display to sleep during certain hours and am not sure the possible ways to do this.  For other use cases on a Raspberry pi, there is a vgencmd command that allows the display to be turned on and off - is there something similar here ?

Thanks.

wmww, 2022-06-11:
Hi! So we don't currently have anything exactly like that, but Ubuntu Frame can dim and turn off the display after some period of inactivity. Perhaps setting a very long timeout would fit your usecase?

To test this out with a timeout of one minute:
```
MIR_SERVER_IDLE_TIMEOUT=60 ubuntu-frame
```

It's not released yet, but the next version will support a Wayland protocol that allows waking up the screen/acquiring a wakelock. A simple command line program could easily be written to wake up the screen at a particular time.

Saviq, 2022-06-12:
Another option is to create a [systemd timer](https://www.freedesktop.org/software/systemd/man/systemd.timer.html) that would change [the display layout](https://github.com/MirServer/ubuntu-frame/blob/master/REFERENCE.md#the-display-snap-configuration-option). So create two layouts, say `default` and `display_off`, and switch between the two.

It does, unfortunately, still require for Frame (and app, by extension) to be restarted, unfortunately.

@wmww's suggestion is probably best, you can already try it with `snap refresh ubuntu-frame --channel edge`. You'd set Frame to go put displays to sleep after a short delay, but use this interface to keep it on while you need it - either from your main application, or from an extra utilty as @wmww suggested:

https://github.com/MirServer/mir/blob/a8f65e4f2cf9a784fe1751b854e3d5a296ce6339/src/wayland/protocol/idle-inhibit-unstable-v1.xml#L27-L83

The advantages here would be that Frame and your app never need restarting, but also that user activity (if applicable) would also wake the screen up.

Saviq, 2022-06-12:
All that said, you should still be able to use `vcgencmd`, as long as you get it on the device in some way.

I think this exhausts what Frame should be doing in this regard, so will close the issue.

closed 2022-06-12 by Saviq