canonical/ubuntu-frame #75: Is there any way to run chromium on Ubuntu Core? I've just installed chromium snap on a device running Ubuntu Core and I wasn't able to run it.

Saviq, 2022-06-20:
Is there any way to run chromium on Ubuntu Core? I've just installed chromium snap on a device running Ubuntu Core and I wasn't able to run it.

`chromium --enable-features=UseOzonePlatform --ozone-platform=wayland
[4361:4361:0620/151041.863045:ERROR:wayland_connection.cc(209)] Failed to connect to Wayland display
[4361:4361:0620/151041.863062:ERROR:ozone_platform_wayland.cc(226)] Failed to initialize Wayland platform
[4361:4361:0620/151041.863067:ERROR:env.cc(226)] The platform failed to initialize.  Exiting.
`

_Originally posted by @jdbelo in https://github.com/MirServer/mir/issues/2463#issuecomment-1160570626_

Saviq, 2022-06-20:
> [4361:4361:0620/151041.863045:ERROR:wayland_connection.cc(209)] Failed to connect to Wayland display

This suggests that the user you are running `chromium` as does not have an available Wayland display.

I'm going to guess you have Frame running as root, and suggest you run `chromium` as root. (I've not tried this, but it should fix the error you are seeing)

_Originally posted by @AlanGriffiths in https://github.com/MirServer/mir/issues/2463#issuecomment-1160577887_

Saviq, 2022-06-20:
I did have the ubuntu-frame snap installed. So I installed and tried to run the same command, and I got some errors.

`sudo chromium --enable-features=UseOzonePlatform --ozone-platform=wayland
[6095:6095:0620/171420.610217:ERROR:zygote_host_impl_linux.cc(90)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.`

After adding --no-sandbox

`sudo chromium --enable-features=UseOzonePlatform --ozone-platform=wayland --no-sandbox
(chrome:6168): Gdk-WARNING **: 17:14:55.105: Settings portal not found: Unknown or unsupported transport “disabled” for address “disabled:”
[6168:6251:0620/171455.174157:ERROR:bus.cc(398)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6168:6251:0620/171455.174195:ERROR:bus.cc(398)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6168:6256:0620/171455.176816:ERROR:udev_watcher.cc(52)] Failed to initialize a udev monitor.
[6168:6251:0620/171455.186089:ERROR:bus.cc(398)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6168:6251:0620/171455.186113:ERROR:bus.cc(398)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6261:6261:0620/171455.212009:ERROR:gpu_init.cc(481)] Passthrough is not supported, GL is egl, ANGLE is 
[6168:6168:0620/171455.219580:ERROR:wayland_event_watcher.cc(36)] libwayland: wl_display@1: error 1: invalid arguments for wl_shm@16.create_pool
Trace/breakpoint trap`

I'd like to reinforce that I'm running these commands on **Ubuntu Core** device.

_Originally posted by @jdbelo in https://github.com/MirServer/mir/issues/2463#issuecomment-1160684000_

Saviq, 2022-06-20:
@jdbelo I've moved this issue here instead.

You'll need to look into `sudo dmesg -w` for any `DENIED` messages as those would highlight confinement issues.

That said, can you explain why do you want to run full Chromium with Ubuntu Frame? It's ill suited to kiosk use cases, https://snapcraft.io/wpe-webkit-mir-kiosk is a much more purpose-built solution.

frank-dspeed, 2022-06-20:
@Saviq this is not true i am from the chromium team i can tell you that chromium is well for that and it even has more.

Maybe some Background for you Chromium is not only a browser it is a whole Platform it can run all kind of apps even in a single window but can easy expand to a multi window solution

we got our own window and display manager inside chromium we even use chromium as the fundamental foundation of chromium os so chrome os so chrome books

Saviq, 2022-06-20:
@frank-dspeed sorry, I conflated Chromium with the browser, as that's what the OP was about.

The use cases of which you write don't have much use of a display server/compositor like Mir/Frame, do they?

frank-dspeed, 2022-06-20:
@Saviq they have as we use that under the hood thats why we support that protocol 

for example you can run it with the --app switch and then give it some positions and some other parameters to run fullscreen without tabs https://peter.sh/experiments/chromium-command-line-switches/

frank-dspeed, 2022-06-20:
@Saviq also there are chromeium apps extensions that we can run fullscreen ubuntu-frame or lets say the MirServer is core tech for chromium apps.

frank-dspeed, 2022-06-21:
I am at present by the way working on the final integration of the chromium platform and this ubuntu frame snap but i will republish it as mir frame to reduce complexity for new coders i my self did not know that ubuntu frame is mir kiosk and so on.

It takes always some time to follow the development of so many projects

frank-dspeed, 2022-06-21:
the ozon layer over Aura allows us also to run on windows and mac with different display server implementations at the same time. so a Chromium Windows Build can still mount a x11 socket for example as long as the Object and API is there.

AlanGriffiths, 2022-06-21:
> [6168:6168:0620/171455.219580:ERROR:wayland_event_watcher.cc(36)] libwayland: wl_display@1: error 1: invalid arguments for wl_shm@16.create_pool Trace/breakpoint trap

This needs `--disable-dev-shm-usage`

AlanGriffiths, 2022-06-21:
@frank-dspeed @jdbelo the `chromium` snap being used here is intended for desktop and while the addition of options may get it to run on Ubuntu Core as an experiment, these configuration options need to be integrated into snap package for serious IoT usage. (I can't speak to whether this would be better added to the desktop snap, or done as a separate snap.)

frank-dspeed, 2022-06-21:
@AlanGriffiths everything needed from ubuntu frame that is useable without Problems i guess everything gets 1:1 mir-frame and then chromium-mir as a snap pre configured to work with mir-frame we need to keep mir frame as single snap as we need to connect 3th party apps also it makes no sense to integrate it directly into the chromium snap

frank-dspeed, 2022-06-21:
later plans are to even split diffrent parts of the chromium platform into individual snaps if that is possible i saw such patterns we will try to take that also into account.

AlanGriffiths, 2022-06-21:
@frank-dspeed I don't (yet) understand the point of introducing `mir-frame`. It sounds the same as Ubuntu Frame (or mir-kiosk) except not supported by Canonical? Does anyone benefit from the confusion caused by another near-identical choice?

Is `chromium-mir` simply a Wayland based kiosk browser? There's nothing Mir specific about that and it should work with other Wayland compositors. Is there any purpose to having Mir in the name?

frank-dspeed, 2022-06-21:
@AlanGriffiths the idea is the following
Only Offering some Launch Options for the integration that will lead later to maybe only a --mir-frame command line switch to set many flags to diffrent defaults. 

I have a meeting next week with other chromium and chrome publishers to create the final strategy 

but the -mir snap should use a chromium base snap and simply run it with defaults as also allow it to more easy run it without to much command line switches.

AlanGriffiths, 2022-06-21:
s/mir socket like the x11 socket/Wayland socket like the x11 socket/

Saviq, 2023-01-27:
I'll close this as an issue and direct to the Ubuntu Discourse that's better suited for broader discussions:

https://discourse.ubuntu.com/c/mir/15

closed 2023-01-27 by Saviq