canonical/ubuntu-frame #77: Add documentation about how to use ubuntu-frame-osk

m2scared, 2022-07-21:
Hi,

I have tried to get `ubuntu-frame-osk` to work unsuccessfully by following https://discourse.ubuntu.com/t/ubuntu-frame-osk-documentation/27568.

The service was crashing with `No virtual keyboard manager Wayland global available` on **release** and **edge** versions.

I was only able to get it working when I found this issue https://github.com/MirServer/mir/issues/2063 which resonates with the problem.

By simply adding the config below `ubuntu-frame-osk` works fine on **edge**.

```
snap set ubuntu-frame config "
add-wayland-extensions=zwlr_layer_shell_v1:zwp_virtual_keyboard_manager_v1:zwp_text_input_manager_v3:zwp_input_method_manager_v2
"
```

ubuntu-frame `71-mir2.8.0+dev398`
ubuntu-frame-osk `40-squeekboard-v1.17.1`
wpe-webkit-mir-kiosk `2.30.5`

AlanGriffiths, 2022-07-21:
Hi, that shouldn't be necessary and I need a bit more information to work out what's happening.

1. What operating system are you using?
2. How are you running Ubuntu Frame and Ubuntu Frame OSK? From the command line? As daemons?

m2scared, 2022-07-22:
Hi @AlanGriffiths,

1. I am using Archlinux minimal console only installation `5.18.12-arch1-1`.
2. Basically I followed the tutorials for installing Ubuntu Frame and Ubuntu Frame OSK. From command line, yes. Using daemons, yes.

```
git clone https://aur.archlinux.org/snapd.git
cd snapd
makepkg -si
sudo systemctl enable --now snapd.socket
sudo snap install ubuntu-frame --edge
sudo snap set ubuntu-frame daemon=true
sudo snap install wpe-webkit-mir-kiosk
sudo snap connect wpe-webkit-mir-kiosk:wayland
sudo snap set wpe-webkit-mir-kiosk daemon=true
sudo snap set wpe-webkit-mir-kiosk url=https://google.com/
sudo snap start wpe-webkit-mir-kiosk
sudo snap install ubuntu-frame-osk --edge
sudo snap connect ubuntu-frame-osk:wayland
sudo snap set ubuntu-frame-osk daemon=true
```

AlanGriffiths, 2022-07-22:
> 1. I am using Archlinux minimal console only installation `5.18.12-arch1-1`.

Thanks @m2scared!

I suspect (it needs investigation) that the AppArmor support is problematic in this environment. A bit of background: Frame uses AppArmor to identify the ubuntu-frame-osk snap and enable the necessary protocols for this client.

The workaround you suggest would likely be needed if AppArmor isn't working right.

----

The investigation:

Could you check the ubuntu-frame logs when the OSK fails to connect (i.e. without overriding the configuration)? If my guess is correct then you should see messages mentioning "aa_getpeercon() failed for process".

You can watch the logs with:

```
snap logs -f ubuntu-frame
```

m2scared, 2022-07-22:
Hi @AlanGriffiths,

Indeed:

```
2022-07-22T08:35:02-03:00 ubuntu-frame.daemon[1997]: [2022-07-22 08:35:02.408902] < - debug - > frame: aa_getpeercon() failed for process 3006: Invalid argument
```

closed 2022-07-28 by bors[bot]

AlanGriffiths, 2022-07-28:
@m2scared the version of Ubuntu Frame now on `beta` ought to work for you without the `add-wayland-extensions` configuration

m2scared, 2022-08-01:
Hi @AlanGriffiths,

I confirm it is working. Thanks.

commit 21ac36d7 (2022-07-28): Identify client without AppArmor (#78)

Fixes: #77
commit 5842b91c (2022-07-28): Identify client without AppArmor (#78)

Fixes: #77
commit 94f2ed2b (2022-07-28): Identify client without AppArmor (#78)

Fixes: #77