canonical/ubuntu-frame #84: RFC: Add a `ubuntu-frame.wake` utility

Saviq, 2022-08-18:
It came up in #68 and also [on #mir-server](https://irclogs.ubuntu.com/2022/08/17/%23mir-server.txt). It could be useful to have a utility to wake the screen up when it powered off due to inactivity.

It could take an optional argument of `--for [seconds]` or so, so you can trivially inhibit idle power off based on a timer.

wmww, 2022-08-18:
It would be nice if this could be a general utility we package into Ubuntu Frame, but could also be used by anyone on any compositor. As I noted in the discussion, one challenge to implementing this is that the protocol requires a surface. I see three solutions:
1. Create a non-mapped surface. This would currently work in Mir, but would go against the spirit of the protocol and might break in the future/on other compositors
2. Create an invisible, non-interactive surface with Layer Shell. This should work anywhere Layer Shell is enabled.  It could also fall back to XDG shell, though then it would steal input focus. It's the most complicated to implement.
3. Propose a revision of the protocol that adds a version of the request without the surface requirement. Compositors could then choose if to honor this request.

RAOF, 2022-08-19:
4. Define a new protocol. `wp_idle_inhibit` is specced for an untrusted cilent, and has all the usual “the compositor may choose to ignore this for any reason” behaviour that implies.
5. Implement `wlr_output_power_management`; this is not *exactly* what we need, but maybe could be bodged into the right shape?
6. Add a non-Wayland control plane to Frame. We're already a dbus client; adding a dbus service should not be particularly difficult.

(6) has the non-obvious benefit that we already have good tooling for confining DBus; my understanding is an ubuntu-frame-control slot/plug would be easy to define. I'm not advocating for this particularly strongly, but I don't think that compositor-wide control is necessarily better provided by Wayland protocol.

PHB-fra, 2024-02-14:
hi,

Possible workaround simulate touch.

apt-get install evemu-tools (on debian / ubuntu)

sudo evemu-record > recording.evemu (For recording the session / mousse, keyboard ...)
Choose the event
Move/key press on physical hardware

on ssh session : 
evemu-play /dev/input/eventX < recording.evemu (for play the session and wake up screen)

Victory the screen wake up !