canonical/ubuntu-frame #88: ubuntu-frame fails to install in Jetson nano

Eldho1416, 2022-09-26:
Hi,

I have been trying to install ubuntu-frame in Jeston and from past week I have been trying many ways but I'm not able to install it in Jetson.

OS: `Ubuntu 18.04.6 LTS (GNU/Linux 4.9.253-tegra aarch64)`

Snap version: `$ snap version
snap    2.57.1
snapd   2.57.1
series  16
ubuntu  18.04
kernel  4.9.253-tegra`

Error: `$ sudo snap install ubuntu-frame
error: cannot perform the following tasks:
- Mount snap "ubuntu-frame" (3901) (systemctl command [start snap-ubuntu\x2dframe-3901.mount] failed with exit status 1: Job for snap-ubuntu\x2dframe-3901.mount failed.
See "systemctl status "snap-ubuntu\\x2dframe-3901.mount"" and "journalctl -xe" for details.
)
- Mount snap "mesa-core20" (126) (systemctl command [start snap-mesa\x2dcore20-126.mount] failed with exit status 1: Job for snap-mesa\x2dcore20-126.mount failed.
See "systemctl status "snap-mesa\\x2dcore20-126.mount"" and "journalctl -xe" for details.
)`

closed 2022-09-26 by Saviq

Eldho1416, 2022-09-26:
Hi @Saviq,

Yes I found the error it is related to kernel `Sep 26 16:38:19 ubuntu kernel: squashfs: SQUASHFS error: Filesystem uses "lzo" compression. This is not supported` and need to fixed by Nvidia, anyways thanks for your input.

Saviq, 2022-09-26:
@Eldho1416 FWIW you can rebuild the snap having removed this line:

https://github.com/MirServer/ubuntu-frame/blob/058551a72bf5abedc87946d6f54f0fd202e7e799/snap/snapcraft.yaml#L8

Eldho1416, 2022-09-26:
Hi @Saviq,

Thank you for your input I will try it out.

AlanGriffiths, 2022-09-27:
> @Eldho1416 FWIW you can rebuild the snap having removed this line:
> 
> https://github.com/MirServer/ubuntu-frame/blob/058551a72bf5abedc87946d6f54f0fd202e7e799/snap/snapcraft.yaml#L8

FWIW mir-kiosk doesn't use `lzo`

Eldho1416, 2022-09-27:
Hi @AlanGriffiths, 

I'm facing issue in installing mesa-core20 and ubuntu-frame not in installing mir-kiosk.

Eldho1416, 2022-09-27:
> @Eldho1416 FWIW you can rebuild the snap having removed this line:
> 
> https://github.com/MirServer/ubuntu-frame/blob/058551a72bf5abedc87946d6f54f0fd202e7e799/snap/snapcraft.yaml#L8

Hi @Saviq,

can you help me understand how to remove this line a rebuild? Just install this repo remove the line from the fine and do `make` or something?

AlanGriffiths, 2022-09-27:
> I'm facing issue in installing mesa-core20 and ubuntu-frame not in installing mir-kiosk.

Yes, but the problem is with the host OS you're running not supporting `lzo`. `mir-kiosk` has similar functionality to Frame and `mesa-core20` but doesn't use `lzo`. That should get you past install problems.

Saviq, 2022-09-27:
> can you help me understand how to remove this line a rebuild? Just install this repo remove the line from the fine and do `make` or something?

Just to answer this: https://snapcraft.io/docs

> That should get you past install problems.

But that, too - you can use `mir-kiosk`, it's the predecessor to Frame.

Eldho1416, 2022-09-27:
Okay Thank you @Saviq and @AlanGriffiths, I'm trying to add squashfs in kernel if it doesn't work I shall try this.