canonical/ubuntu-frame #92: ubuntu frame daemon fails

Eldho1416, 2022-10-21:
Hi,

After installing ubuntu-frame in jetson when i hit `sudo snap set ubuntu-frame daemon=true` I am getting below errors, what can i do about this? 

```
2022-10-21T20:04:08+05:30 systemd[1]: Started Service for snap application ubuntu-frame.daemon.
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: + exec nohup /snap/ubuntu-frame/3901/bin/run-frame /snap/ubuntu-frame/3901/usr/local/bin/frame
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: + [ /snap/ubuntu-frame/3901/usr/local/bin/frame = --help ]
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: + dirname /run/user/0/snap.ubuntu-frame
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: + XDG_RUNTIME_DIR=/run/user/0
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: + export XDG_RUNTIME_DIR
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: + mkdir -p /run/user/0 -m 700
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: + [ -n  ]
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: + exec /snap/ubuntu-frame/3901/usr/local/bin/frame
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.185903] <information> mirserver: Starting
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.186898] < - debug - > mirserver: Using Linux VT subsystem for session management
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.187498] <information> mircommon: Loading modules from: /snap/ubuntu-frame/3901/usr/lib/aarch64-linux-gnu/mir/server-platform
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.187651] <information> mircommon: Loading module: /snap/ubuntu-frame/3901/usr/lib/aarch64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.20
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.187710] <information> mircommon: Loading module: /snap/ubuntu-frame/3901/usr/lib/aarch64-linux-gnu/mir/server-platform/server-x11.so.20
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.187749] <information> mircommon: Loading module: /snap/ubuntu-frame/3901/usr/lib/aarch64-linux-gnu/mir/server-platform/input-evdev.so.8
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.190263] <information> mirserver: Found display driver: mir:gbm-kms (version 2.9.0)
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.190898] <information> gbm-kms: Unsupported: No DRM devices detected
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.190951] <information> mirserver: (Unsupported by system environment)
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.190984] <information> mirserver: Found display driver: mir:x11 (version 2.9.0)
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.191044] <information> mirserver: (Unsupported by system environment)
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: ERROR: /build/mir-p5QAkY/mir-2.9.0/src/server/graphics/default_configuration.cpp(233): Throw in function virtual const std::vector<std::shared_ptr<mir::graphics::DisplayPlatform> >& mir::DefaultServerConfiguration::the_display_platforms()
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: Dynamic exception type: boost::wrapexcept<std::runtime_error>
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: std::exception::what: Exception while creating graphics platform
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: ERROR: /build/mir-p5QAkY/mir-2.9.0/src/server/graphics/platform_probe.cpp(190): Throw in function std::vector<std::pair<mir::graphics::SupportedDevice, std::shared_ptr<mir::SharedLibrary> > > {anonymous}::modules_for_device({anonymous}::ModuleType, const std::vector<std::shared_ptr<mir::SharedLibrary> >&, const mir::options::ProgramOption&, const std::shared_ptr<mir::ConsoleServices>&)
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: Dynamic exception type: boost::wrapexcept<std::runtime_error>
2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: std::exception::what: Failed to find any platforms for current system
2022-10-21T20:04:09+05:30 systemd[1]: snap.ubuntu-frame.daemon.service: Main process exited, code=exited, status=1/FAILURE
2022-10-21T20:04:09+05:30 systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'exit-code'.
```

AlanGriffiths, 2022-10-21:
> ```
> 2022-10-21T20:04:09+05:30 ubuntu-frame.daemon[16641]: [2022-10-21 20:04:09.190263] <information> mirserver: Found display driver: mir:gbm-kms (version 2.9.0)
> 2022-10-21T20:04:09+05:30 [...]

This is the essential problem: there are no suitable gbm-kms drivers in the kernel you are using. 
 
The requirements Frame has of the host OS are discussed here: https://mir-server.io/docs/where-does-ubuntu-frame-work

I don't have any Jetson kit to try, but I believe Nvidia are working on gbm-kms support: to use this you would need gbm-kms support in the kernel and a corresponding implementation of the `graphics-core20` content interface to provide Frame with the corresponding userspace drivers.

Eldho1416, 2022-10-21:
Hi @AlanGriffiths,

Hi have learned how to customise Jetson kernel, do you think that will help me here?

AlanGriffiths, 2022-10-21:
> Hi have learned how to customise Jetson kernel, do you think that will help me here?

Maybe, I can't really tell you any more than the above

Eldho1416, 2022-10-21:
Okay thanks for your time

closed 2022-10-21 by Eldho1416