canonical/ubuntu-frame #94: `--diagnostic-path` crashes when existing on start, but removed

Saviq, 2022-10-21:
```
$ touch foo
$ ./build-Debug/frame --diagnostic-path $PWD/foo
# ...
# remove the file
# ...
!!! Fatal signal received. Attempting cleanup, but deadlock may occur
Mir fatal error: Unsupported attempt to continue after a fatal signal: SIGSEGV
!!! Fatal signal received. Attempting cleanup, but deadlock may occur
Mir fatal error: Unsupported attempt to continue after a fatal signal: SIGABRT
[1]    661279 IOT instruction (core dumped)  ./build-Debug/frame --diagnostic-path $PWD/foo
```

graysonguarino, 2022-10-25:
I can't seem to replicate this issue when following these instructions exactly on the newest Mir and Frame builds (building from the `add-crash-reporter-delay` branch). If the issue persists for you on the newest builds, can I get a backtrace?

graysonguarino, 2022-10-25:
Aha! I think I found what may have caused this. Did you happen to have a directory `$PWD/foo`?

I got a similar crash when I did not specify the file's name and just pointed the diagnostic path at a directory.

AlanGriffiths, 2022-10-28:
@Saviq can you check that you can reproduce this? Neither @graysonguarino or I can

Saviq, 2022-10-31:
> @Saviq can you check that you can reproduce this? Neither @graysonguarino or I can

Not any more I can't.

closed 2022-10-31 by AlanGriffiths