canonical/ubuntu-frame #118: Multi-Display Support

allebone, 2022-12-17:
Is there an easy way to support Multi-display support on a single Hardware Client. For example, an HPE t740 Thin Client. It has 4 independent Display Port Outputs, and enough horsepower to visualize the Web Instances we would like to target. 

The goal is 4 completely different web targets, executed together, across 4 screens.

AlanGriffiths, 2022-12-17:
Yes, with the `display` configuration Ubuntu Frame can be configured so that the screens do not overlap (which is the default).

In that configuration, successive windows will be placed on different displays.

See: https://mir-server.io/docs/ubuntu-frame-configuration-options

m2scared, 2022-12-28:
@AlanGriffiths is it possible to run a different `wpe-webkit-mir-kiosk` on each display?

I would like to have each display showing a different page.

AlanGriffiths, 2022-12-28:
> @AlanGriffiths is it possible to run a different `wpe-webkit-mir-kiosk` on each display?

That depends on the `wpe-webkit-mir-kiosk` snap, not Ubuntu Frame. It may be possible to use snapd's "experimental" parallel install feature, but not all snaps work as expected with this.

> I would like to have each display showing a different page.
> 
> I tried to set `group` parameter but after that I am a bit lost. 

I don't see why you would be setting the "group" parameter. That relates to using multiple physical outputs to display one window. (The opposite of what I understand you to want.)

> I can see the same `wpe-webkit-mir-kiosk` on all three monitors

Yes, the default behaviour of Ubuntu Frame is to overlap all the outputs so that the same content appears on each. See my comment above for the configuration options that override this.

m2scared, 2022-12-29:
Thanks @AlanGriffiths.

Still I am not sure how to implement this. Would it be too much to ask for a `demo`?

For instance, suppose I have a snap `background-screen` in which I can set the `background-screen.config.color` to either `red`, `green` or `blue`.

How can one set `background-screen.config.color = red` at `display = 0` ,  `background-screen.config.color = green` at `display = 1`  and `background-screen.config.color = blue` at `display = 2` ?

I tried with:

```bash
snap set ubuntu-frame display="
layouts:
  default:
    cards:
    - card-id: 0
      unknown-1:
        orientation: inverted
"
```

But I was not able to work it out from here.

allebone, 2023-01-01:
@m2scared That along the lines of my original question. I have several Kiosk Browser Instances to display from a single HW Box to 4 Seperate Screens via 4 Display Ports/Cables.

AlanGriffiths, 2023-01-04:
https://mir-server.io/docs/how-to-configure-ubuntu-frame-for-multiple-outputs

allebone, 2023-01-07:
Without any offense...how long has that been there? I swear I looked everywhere first!

AlanGriffiths, 2023-01-07:
About 3 days. It was written in response to feedback like this thread

allebone, 2023-01-09:
Oh! Thats all good news to me! Just making sure I did my part and attempted everything I could before bother you fine folks! I look forward to trying it out!

closed 2023-01-09 by allebone