canonical/ubuntu-frame #123: Hiding cursor doesn't work?

practical-mechanics, 2023-01-26:
Hello,
I'm running Ubuntu Server 22.10 on a PC
I installed **Ubuntu-Frame** and **wpe-webkit-mir-kiosk** and everything is fine, but I can't hide the cursor.
I tried changing to **null**, but the cursor remains in the top left corner.
I'm using a touch screen, and when I tap on it, the cursor does not move (although it is registering the tap).
When restarting **Ubuntu-Frame**, the cursor disappears, but when tapping on the screen, it returns to the corner and remains there.

Am I missing something here?

AlanGriffiths, 2023-01-26:
```
snap set ubuntu-frame config="cursor=null"
```
works for me

Saviq, 2023-01-27:
Confirmed, I'm not seeing the problem reported. @practical-mechanics if you still see this, can you please confirm what your `snap get ubuntu-frame` looks like?

closed 2023-01-27 by Saviq

practical-mechanics, 2023-01-27:
> ```
> snap set ubuntu-frame config="cursor=null"
> ```

This one works,
I tried:
`snap set ubuntu-frame cursor=none`
Thank you!

practical-mechanics, 2023-01-27:
I saw it somewhere on the internet, I guess it's not updated.
Also saw [this thread](https://discourse.ubuntu.com/t/none-is-wrong-for-ubuntu-frame-cursor-configuration/28604) about the "null" and tried it as well but without the word "config".

AlanGriffiths, 2023-01-27:
> I tried:
> `snap set ubuntu-frame cursor=none`
> Thank you!

ubuntu-frame doesn't have a `cursor` option. But, unfortunately, there's no good way to validate that snap configuration options being set are supported by the snap, so we can't give you an error for that.

There's a reference for the supported configuration options:

https://mir-server.io/docs/ubuntu-frame-configuration-options