canonical/ubuntu-frame #128: ubuntu-frame doesn't work with a DisplayLink

edaquinta, 2023-02-23:
I cannot get to work with a DisplayLink display, It worked fine with an HDMI monitor on the same computer, I tried with ubuntu core, server and desktop and same results. 

You can find below the logs if that helps or if I'm missing something. 

Thanks in advance.

******************

sudo snap logs -n 100 ubuntu-frame
[sudo] password for edaquinta: 
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.911573] <information> mircommon: Loading modules from: /snap/ubuntu-frame/4719/usr/lib/x86_64-linux-gnu/mir/server-platform
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.911630] <information> mircommon: Loading module: /snap/ubuntu-frame/4719/usr/lib/x86_64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.20
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.911656] <information> mircommon: Loading module: /snap/ubuntu-frame/4719/usr/lib/x86_64-linux-gnu/mir/server-platform/server-x11.so.20
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.911671] <information> mircommon: Loading module: /snap/ubuntu-frame/4719/usr/lib/x86_64-linux-gnu/mir/server-platform/input-evdev.so.8
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.912440] <information> mirserver: Found rendering driver: mir:gbm-kms (version 2.12.0)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.914279] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card0, driver i915
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.914453] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.914559] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.914663] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.914766] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.914869] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.914961] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.915080] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/renderD128, driver i915
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: Failed to create /root/.cache/mesa_shader_cache for shader cache (Permission denied)---disabling.
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.920713] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card1, driver udl
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: MESA-LOADER: failed to open udl: /snap/ubuntu-frame/4719/graphics/dri/udl_dri.so: cannot open shared object file: No such file or directory (search paths /snap/ubuntu-frame/4719/graphics/dri, suffix _dri)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: failed to load driver: udl
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: Failed to create /root/.cache/mesa_shader_cache for shader cache (Permission denied)---disabling.
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.925583] <information> gbm-kms: Detected software renderer: llvmpipe (LLVM 12.0.0, 256 bits)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926089] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926198] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card2, driver evdi
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926208] <information> gbm-kms: Quirks: skipping device /dev/dri/card2 (matches driver quirk evdi)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926214] <information> gbm-kms: Not probing device /dev/dri/card2 due to specified quirk
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926295] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926379] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card3, driver evdi
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926387] <information> gbm-kms: Quirks: skipping device /dev/dri/card3 (matches driver quirk evdi)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926392] <information> gbm-kms: Not probing device /dev/dri/card3 due to specified quirk
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926467] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926551] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card4, driver evdi
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926558] <information> gbm-kms: Quirks: skipping device /dev/dri/card4 (matches driver quirk evdi)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926563] <information> gbm-kms: Not probing device /dev/dri/card4 due to specified quirk
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926639] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926722] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card5, driver evdi
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926730] <information> gbm-kms: Quirks: skipping device /dev/dri/card5 (matches driver quirk evdi)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926735] <information> gbm-kms: Not probing device /dev/dri/card5 due to specified quirk
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926806] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926871] <information> mirserver: Driver supports:
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926880] <information> mirserver: 	/devices/pci0000:00/0000:00:02.0/drm/renderD128 (priority 256)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926885] <information> mirserver: 	/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.2/1-4.2:1.0/drm/card1 (priority 128)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926894] <information> mirserver: Found rendering driver: mir:x11 (version 2.12.0)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926904] <information> mirserver: (Unsupported by system environment)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926950] <information> mirserver: Selected rendering driver: mir:gbm-kms (version 2.12.0)
2023-02-22T23:21:03-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:03.926977] <information> mirserver: Selected rendering driver: mir:gbm-kms (version 2.12.0)
2023-02-22T23:21:04-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:04.147862] < -warning- > gbm-kms: Unable to determine the current display mode.
2023-02-22T23:21:04-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:04.149940] <information> gbm-kms: DRM device details:
2023-02-22T23:21:04-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:04.149971] <information> gbm-kms: /dev/dri/card0: using driver i915 [Intel Graphics] (version: 1.6.0 driver date: 20201103)
2023-02-22T23:21:04-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:04.150017] <information> gbm-kms: 	Output: DisplayPort-1 (disconnected)
2023-02-22T23:21:04-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:04.161929] <information> gbm-kms: 	Output: HDMI-A-1 (disconnected)
2023-02-22T23:21:04-05:00 ubuntu-frame.daemon[4162]: [2023-02-22 23:21:04.161966] <information> gbm-kms: 	Output: DisplayPort-2 (disconnected)
[... cut ...]
2023-02-22T23:21:04-05:00 ubuntu-frame.daemon[4162]: failed to load driver: udl
2023-02-22T23:21:04-05:00 ubuntu-frame.daemon[4162]: Failed to create /root/.cache/mesa_shader_cache for shader cache (Permission denied)---disabling.
2023-02-22T23:21:04-05:00 systemd[1]: snap.ubuntu-frame.daemon.service: Main process exited, code=dumped, status=11/SEGV
2023-02-22T23:21:04-05:00 systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'core-dump'.
2023-02-22T23:21:04-05:00 systemd[1]: Stopped Service for snap application ubuntu-frame.daemon.

RAOF, 2023-02-23:
This is https://github.com/MirServer/mir/issues/2678 - due to a quirk of the current Mir implementation we (unnecessarily) require that each output of associated with a GL interface that can render to it.

This limitation is in the current batch of work, but we do not yet have an ETA for when the fix will land.

edaquinta, 2023-02-23:
Thanks for the quick response. I will follow the issue

Saviq, 2023-02-23:
Duplicate of https://github.com/MirServer/mir/issues/2678

closed 2023-02-23 by Saviq