canonical/ubuntu-frame #130: Latest Version 95-mir2.12.1 released on 8 March 2023. just bricked all my customer tv displays

cluemaster-robert, 2023-03-10:
The latest version of 95-mir2.12.1 was released on 8 March 2023.

We have been using Ubuntu-Frame for a few years now and everything has been working great. Today all of our customers called our support lines saying that our devices with our TV display software using ubuntu-frame are now booting up and just showing a grey screen forever. Our snap application that has always worked is no longer working at all or starting.  We were able to ssh into our test devices and as soon as we login via ssh, our application would launch and start working. If we reboot the device it would return to the ubuntu-frame grey screen and stay there forever.   We downgraded by using a `snap refresh ubuntu-frame --channel=20/latest`  which downgraded and installed 92-mir2.12.0 and our application immediately started displaying after it was installed and the service started.  Rebooting the PC also was normal and would fully launch the ubuntu-frame and connect our application for video.

Something in yesterday's latest update to Ubuntu-Frame either has a bug or some major change that is causing snaps like ours to no longer work at all.

Saviq, 2023-03-10:
Hi @cluemaster-robert sorry about that, we must have missed your case in our testing (of which there's plenty, I guarantee you).

First, you can [`snap revert <snap>`](https://snapcraft.io/docs/getting-started#heading--revert) any time you encounter any problem with an upgrade - it will go back to the previous revision.

I've now rolled back the release, so a manual `snap refresh`, or a reboot or a wait until the next automatic refresh will bring the previous revision in. That should bring your customers back (and not hurt any more, if they didn't yet receive the update).

Now. To help debug what's gone wrong, may I ask that you attach the logs from both Ubuntu Frame and your application (e.g. [`snap logs -n500 <snap>`](https://snapcraft.io/docs/service-management#heading--logs)).

You should subscribe to [the Mir discourse category](https://discourse.ubuntu.com/c/mir/15) in which we announce a call for testing for each Frame release (e.g. [the last one](https://discourse.ubuntu.com/t/call-for-testing-mir-kiosk-mir-test-tools-egmde-ubuntu-frame-ubuntu-frame-vnc/34270)). You also should run your testing devices on the `candidate` channel, in which we keep our next stable for at least a week during the call for testing.

When we understand your issue, we'll try and include your case in our testing. If you want official support and our QA pipeline replicate your exact setup, please contact our support team at https://ubuntu.com/support.

AlanGriffiths, 2023-03-10:
I think @Saviq has tracked this down to a [problem with wpe-webkit-mir-kiosk](https://gitlab.com/glancr/wpe-webkit-snap/-/issues/49).

@cluemaster-robert, could you confirm you are using wpe-webkit-mir-kiosk as your application snap?

Saviq, 2023-03-10:
It could also be any other snap based off of the deprecated https://github.com/MirServer/mir-kiosk-snap-launch/

Saviq, 2023-03-10:
@cluemaster-robert we will be releasing a version of Frame that includes a workaround for this issue (see #131). You can try the `beta` channel today, I'll be pushing it to `candidate` for another call-for-testing period. It would be great if you could feed back whether it's fixing things for you.

cluemaster-robert, 2023-03-10:
Good Morning.

Thank you, Thank you, Thank you!!!.   I really appreciate the quick
response to this and for you reverting your version to the previous one.
Were in the PST time so late last night we were drafting up emails to send
to our customers today, that they might need to ship back our display
devices to us. This helps us and all others using Ubuntu-kiosk to find the
issue so we can build new apps or whatnot. Thank you.   We're telling our
customers to have turned on the devices in the past two days, to just wait
this morning and leave it on the grey screen and about 4 hours later it
should automatically update and start displaying.

I'm subscribed to the Mir Discourse Category now. Thank you for reminding
me about that. We will watch for the posts more closely here when new
versions come out.

We're taking a test device and manually refreshing to the
--channel=22/latest.  It's making us install "mesa-core22-test" which I did
not do when that broken version was released. Hopefully, that had nothing
to do with it.  So instead since versions have moved around a bit in the
snap store were installing --channel=latest/beta

What's strange is the Web Snap store shows this

[image: image.png]

But it's forcing the install on our test VM to be "97-mir2.12.1".  But it's
working!  My test VM is starting up on its own and WITHOUT logging into it
with SSH, our snap is launching and connecting to ubuntu-frame with no
issues. So maybe something specific in 95xx.21.1 broke something inside,
because it would not allow anything to display until you ssh into the box
itself.  But I have rebooted many many times and our device snap is booting
up and connecting to ubuntu-frame with no problem. It's working well on
this version.

[image: image.png]

I'm unable to get the logs that you requested because I can't seem to
install the broken version 95xx.21.1.

Robert Bartram
CEO, Founder
ClueMaster.io
Escape Room Management

Saviq, 2023-03-10:
@cluemaster-robert not `22/`, that's a work in progress and will not be stable for some time still.

Please stick to `20/stable` for now - and we'll announce on Discourse when things are changing in that regard.

cluemaster-robert, 2023-03-10:
My snap dev says he did not use that.  Here is our yaml file we built from
last yaer.

https://github.com/ClueMaster-LLC/MediaDisplay-MPV/blob/main/cluemaster-mediadisplay-core/snap/snapcraft.yaml

Robert Bartram
CEO, Founder
ClueMaster.io
Escape Room Management

cluemaster-robert, 2023-03-10:
Hey Alan.  Thanks for helping to figure this out.   Here is our yaml file.
https://github.com/ClueMaster-LLC/MediaDisplay-MPV/blob/main/cluemaster-mediadisplay-core/snap/snapcraft.yaml

Robert Bartram
CEO, Founder
ClueMaster.io
Escape Room Management

Saviq, 2023-03-10:
@cluemaster-robert we don't see that, it's private. Please can you test `20/beta` or `latest/beta`, if the workaround works for you? If it doesn't, please review https://mir-server.io/docs/developer-guide for how we recommend setting things up.

cluemaster-robert, 2023-03-10:
> @cluemaster-robert we don't see that, it's private. Please can you test `20/beta` or `latest/beta`, if the workaround works for you? If it doesn't, please review https://mir-server.io/docs/developer-guide for how we recommend setting things up.

Sorry about that. I forgot we had our git on private.  Here is our .yaml file.

Also..  how do you tell what channel it's on when it installs?   I just see the version when you "snap list". 
It's working well on 20/latest, and also whatever is 97-mir2.12.1

[image]

name: cluemaster-mediadisplay-core 
base: core20 
version: '1.0.23'
summary: Clumaster.io Media Display for Escape Rooms 
description: |
  ClueMaster Media Display is a Dynamic TV Display software for escape rooms.
  It shows escape room countdown timers, visual text messages, video, audio,
  and photo clues.

grade: stable 
confinement: strict 

apps:
  cluemaster-mediadisplay-core:
    daemon: simple
    restart-condition: always
    command-chain:
      - env-setup
    command: usr/local/bin/x11_kiosk_launch $SNAP/bin/desktop-launch $SNAP/bin/prepare-launch $SNAP/cluemaster_display
    environment:
      DISABLE_WAYLAND: 1

  pulseaudio:
    command: bin/pulseaudio
    daemon: simple
    restart-condition: always
    install-mode: enable

  pactl:
    command-chain: [bin/client-wrapper]
    command: usr/bin/pactl

  paplay:
    command-chain: [bin/client-wrapper]
    command: usr/bin/paplay

  parec:
    command-chain: [bin/client-wrapper]
    command: usr/bin/parec

  config:
    command: bin/config

plugs:
  playback:
    interface: audio-playback
  wayland:
  network:
  opengl:
  network-bind:
  shutdown:
  hardware-observe:
  alsa:

slots:
  audio-playback:

parts:
  copy-source-code:
    after: [desktop-qt5]
    plugin: dump
[... 283 lines cut ...]
    symlink: $SNAP/usr/bin/xkbcomp
  /usr/share/icons:
    bind: $SNAP/usr/share/icons
  /usr/share/fonts:
    bind: $SNAP/usr/share/fonts
  /usr/local/share/fonts:
    bind: $SNAP/usr/local/share/fonts
  /etc/fonts:
    bind: $SNAP/etc/fonts
  /etc/pulse:
    bind: $SNAP/etc/pulse
  /etc/alsa:
    bind: $SNAP/etc/alsa
  /var/lib/pulse:
    bind: $SNAP_DATA
  /usr/lib/pulse-15.0:
    symlink: $SNAP/usr/lib/pulse-15.0
  /usr/lib/alsa-lib:
    bind: $SNAP/usr/lib/alsa-lib
  /usr/share/pulseaudio:
    symlink: $SNAP/usr/share/pulseaudio
  /usr/share/alsa:
    symlink: $SNAP/usr/share/alsa
  /usr/share/applications:
    bind: $SNAP/usr/share/applications

AlanGriffiths, 2023-03-10:
OK, I see. You are using a different old approach (with https://github.com/MirServer/mir-kiosk-x11-snap) with the same buggy `wait_for()` function. The same workaround in Frame should work (and you have confirmed it does).

However, you may want to update your packaging to a more recent approach by following one of the following:

https://mir-server.io/docs/packaging-a-qt5-application-as-an-iot-gui

https://mir-server.io/docs/packaging-an-x11-application-as-an-iot-gui

Thanks for bringing this to our attention.

cluemaster-robert, 2023-03-11:
Thank you for confirming and seeing where the buggy issue is. My dev wants
to bake the Ubuntu-frame into our snap instead of having us install it
manually at the OS level, just so it can only be updated when we build new
space and not update on its own.  Would you recommend that as well?

Robert

Saviq, 2023-03-12:
> Also.. how do you tell what channel it's on when it installs?

```yaml
$ snap info ubuntu-frame
# ...
tracking:     latest/edge
#...
```

> My dev wants to bake the Ubuntu-frame into our snap instead of having us install it manually at the OS level, just so it can only be updated when we build a new snap and not auto update on its own.

It would mean you have to take over maintenance of it, including reacting to security notices. It's a heavy hammer, you're free to do it - our recommendation would certainly be to follow `stable`, with your testing setup on `beta`, or even `edge` to notice any problems quicker.

In any case, have a look at our updates whitepaper https://ubuntu.com/engage/snapd-whitepaper for best practices around fleet management.

AlanGriffiths, 2023-03-12:
@cluemaster-robert remember the problem you encountered was from a bug in software you already pulled into your snap. Specifically the buggy `wait_for()` function. (See below for the details)

In our current snaps and examples, we've encountered, identified and fixed this bug. Your snap still has the buggy version and consequently you encountered the problem.

 The problem wasn't the Frame changing, it was code in your snap not being maintained. If you were to "bake ubuntu-frame" into your snap, it wouldn't make you immune to bugs in the code. But it would give more opportunities for bugs that are fixed elsewhere to remain in your snap.

----

The bug explained:

```bash
wait_for()
{
  until
    inotifywait --event create "$(dirname "$1")"&
    inotify_pid=$!
    [ -O "$1" ]
  do
    wait "${inotify_pid}"
  done
  kill "${inotify_pid}"
}
```
The problem isn't obvious: It is a race between `inotifywait` creating a watch, the ` [ -O "$1" ]` test for existence and the creation of the file. It is possible for the existence test to execute first, then the create and finally the watch. After that the function blocks.

This is an accident waiting to happen: the Frame release just happened to be the change that led to that relative timing between your snap and Frame starting. But it could equally have been an update to a base snap, the kernel snap or running on different hardware.

AlanGriffiths, 2023-03-13:
@Saviq I have updated the [mir-kiosk-x11](https://snapcraft.io/mir-kiosk-x11) snap with the same fix:

https://github.com/MirServer/mir-kiosk-x11-snap/commit/15eeb947a67ea48587fa194aae0c47caea475021

@cluemaster-robert that means you will pick up the corrected code next time you rebuild your snap.

cluemaster-robert, 2023-03-13:
Thank you so much. We will try to better testing and keep our main snap
updated more often.

Robert

commit 7c5f8525 (2023-03-10): Unblock clients with a racy `wayland-launch` script (#131)

Fixes: #130