canonical/ubuntu-frame #134: Keyboard appearing at top of screen

jaydensmith, 2023-04-04:
Seemingly at random (it was working for months up until now) the keyboard has started appearing at the top of the screen, overlaying part of the application. Sometimes a blank space will show at the bottom of the screen where the keyboard should be, but not always.

[image]
[image]

Nothing in the logs seems to lend a clue:
```
2023-04-04T00:47:52Z systemd[1]: Started Service for snap application ubuntu-frame-osk.daemon.
2023-04-04T00:48:05Z ubuntu-frame-osk.daemon[1312]: Setting up watches.
2023-04-04T00:48:05Z ubuntu-frame-osk.daemon[1312]: Watches established.
2023-04-04T00:48:05Z ubuntu-frame-osk.daemon[1312]: /run/user/ CREATE,ISDIR 0
2023-04-04T00:48:05Z ubuntu-frame-osk.daemon[1326]: Setting up watches.
2023-04-04T00:48:05Z ubuntu-frame-osk.daemon[1326]: Watches established.
2023-04-04T00:48:06Z ubuntu-frame-osk.daemon[1326]: /run/user/0/ CREATE wayland-0.lock
2023-04-04T00:48:06Z ubuntu-frame-osk.daemon[1014]: Setting input sources to [('xkb', 'us')]
2023-04-04T00:48:07Z ubuntu-frame-osk.daemon[1014]: Debug: Tried file "/root/snap/ubuntu-frame-osk/193/.local/share/squeekboard/keyboards/us.yaml", but it's missing: No such file or directory (os error 2)
2023-04-04T00:48:07Z ubuntu-frame-osk.daemon[1014]: Info: Loaded layout Resource: us
2023-04-04T00:48:07Z ubuntu-frame-osk.daemon[1014]: Debug: Tried file "/root/snap/ubuntu-frame-osk/193/.local/share/squeekboard/keyboards/us.yaml", but it's missing: No such file or directory (os error 2)
2023-04-04T00:48:07Z ubuntu-frame-osk.daemon[1014]: Info: Loaded layout Resource: us
2023-04-04T00:48:07Z -[1014]: Could not connect to session manager: Cannot autolaunch D-Bus without X11 $DISPLAY

2023-04-04T00:48:07Z -[1014]: Failed to init libfeedback: Cannot autolaunch D-Bus without X11 $DISPLAY
2023-04-04T02:14:39Z ubuntu-frame-osk.daemon[1014]: Can't connect to the bus: Cannot autolaunch D-Bus without X11 $DISPLAY. Visibility switching unavailable.Debug: Tried file "/root/snap/ubuntu-frame-osk/193/.local/share/squeekboard/keyboards/us.yaml", but it's missing: No such file or directory (os error 2)
2023-04-04T02:14:39Z ubuntu-frame-osk.daemon[1014]: Info: Loaded layout Resource: us
2023-04-04T02:14:39Z ubuntu-frame-osk.daemon[1014]: Debug: Tried file "/root/snap/ubuntu-frame-osk/193/.local/share/squeekboard/keyboards/us_wide.yaml", but it's missing: No such file or directory (os error 2)
2023-04-04T02:14:39Z ubuntu-frame-osk.daemon[1014]: Info: Loaded layout Resource: us_wide
```

AlanGriffiths, 2023-04-04:
> ubuntu-frame `98-mir2.12.1`

Thanks, that confirms you have #132 (which fixed the problem we saw)

AlanGriffiths, 2023-04-04:
Reproduced

AlanGriffiths, 2023-04-04:
> > ``
> 
> Thanks, that confirms you have #132 (which fixed the problem we saw)

Actually, I was wrong there. We have a mismatch between our ongoing work to support core22 and the `latest` and `20` tracks. I've applied this fix, but we need to wait for the build infrastructure to deploy the changes.

Once that is in place I'll promote them to `candidate` for testing

AlanGriffiths, 2023-04-04:
I have promoted the corrected Frame to `candidate` and [noted this on the call for testing](https://discourse.ubuntu.com/t/call-for-testing-ubuntu-frame-mir-test-tools-mir-kiosk-egmde-mir-2-13-0-update/34812/4).

@jaydensmith thanks for bringing this to our attention.

Saviq, 2023-04-06:
This is now in stable across the board.

closed 2023-04-06 by Saviq