canonical/ubuntu-frame #142: Ubuntu-Frame Screen Blinking

NDTechSupport, 2023-06-21:
I'm having an issue with ubuntu frame blinking/crashing on an ubuntu core device when screen layout is set to default and using the built in and an external display. snap set ubuntu-frame display-layout=side_by_side works, but default, to mirror the displays, does not. Here's what the ubuntu-frame logs say:

`2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.081567] <information> GLRenderer: GL max texture size = 16384
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.081771] <information> GLRenderer: GL framebuffer bits: RGBA=8880, depth=0, stencil=0
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.082513] < - ERROR - > gbm-kms: Output DisplayPort-2 has no associated CRTC to schedule page flips on
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.092952] <information> mirserver: Initial display configuration:
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093060] <information> mirserver: * Output 1: eDP connected, used
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093091] <information> mirserver: . |_ EDID manufacturer: CMN
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093109] <information> mirserver: . |_ EDID product code: 4409
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093132] <information> mirserver: . |_ Physical size 11.6" 260x140mm
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093146] <information> mirserver: . |_ Power is on
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093162] <information> mirserver: . |_ Current mode 1366x768 60.00Hz
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093177] <information> mirserver: . |_ Preferred mode 1366x768 60.00Hz
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093190] <information> mirserver: . |_ Orientation normal
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093207] <information> mirserver: . |_ Logical size 1366x768
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093221] <information> mirserver: . |_ Logical position +0+0
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093233] <information> mirserver: . |_ Scaling factor: 1.00
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093246] <information> mirserver: * Output 2: DisplayPort disconnected
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093258] <information> mirserver: * Output 3: DisplayPort connected, used
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093271] <information> mirserver: . |_ EDID monitor name: LG TV
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093283] <information> mirserver: . |_ EDID manufacturer: GSM
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093295] <information> mirserver: . |_ EDID product code: 1
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093309] <information> mirserver: . |_ Physical size 72.3" 1600x900mm
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093321] <information> mirserver: . |_ Power is on
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093335] <information> mirserver: . |_ Current mode 1920x1080 59.94Hz
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093349] <information> mirserver: . |_ Preferred mode 1920x1080 60.00Hz
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093362] <information> mirserver: . |_ Orientation normal
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093375] <information> mirserver: . |_ Logical size 1920x1080
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093387] <information> mirserver: . |_ Logical position +0+0
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: [2023-06-21 19:10:03.093399] <information> mirserver: . |_ Scaling factor: 1.00
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: Mir fatal error: Output DisplayPort-2 has no associated CRTC to wait on
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: !!! Fatal signal received. Attempting cleanup, but deadlock may occur
2023-06-21T19:10:03Z ubuntu-frame.daemon[5230]: Mir fatal error: Unsupported attempt to continue after a fatal signal: SIGABRT
2023-06-21T19:10:03Z systemd[1]: snap.ubuntu-frame.daemon.service: Main process exited, code=killed, status=6/ABRT
2023-06-21T19:10:03Z systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'signal'.`

Saviq, 2023-06-21:
Hi @NDTechSupport, can you please attach output from `drm_info`? It's also packaged in [graphics-test-tools](https://snapcraft.io/graphics-test-tools/)

NDTechSupport, 2023-06-21:
`Node: /dev/dri/card0
├───Driver: amdgpu (AMD GPU) version 3.42.0 (20150101)
│   ├───DRM_CLIENT_CAP_STEREO_3D supported
│   ├───DRM_CLIENT_CAP_UNIVERSAL_PLANES supported
│   ├───DRM_CLIENT_CAP_ATOMIC supported
│   ├───DRM_CLIENT_CAP_ASPECT_RATIO supported
│   ├───DRM_CLIENT_CAP_WRITEBACK_CONNECTORS supported
│   ├───DRM_CAP_DUMB_BUFFER = 1
│   ├───DRM_CAP_VBLANK_HIGH_CRTC = 1
│   ├───DRM_CAP_DUMB_PREFERRED_DEPTH = 24
│   ├───DRM_CAP_DUMB_PREFER_SHADOW = 1
│   ├───DRM_CAP_PRIME = 3
│   ├───DRM_CAP_TIMESTAMP_MONOTONIC = 1
│   ├───DRM_CAP_ASYNC_PAGE_FLIP = 1
│   ├───DRM_CAP_CURSOR_WIDTH = 128
│   ├───DRM_CAP_CURSOR_HEIGHT = 128
│   ├───DRM_CAP_ADDFB2_MODIFIERS = 0
│   ├───DRM_CAP_PAGE_FLIP_TARGET = 0
│   ├───DRM_CAP_CRTC_IN_VBLANK_EVENT = 1
│   ├───DRM_CAP_SYNCOBJ = 1
│   └───DRM_CAP_SYNCOBJ_TIMELINE = 1
├───Device: PCI 1002:98e4
├───Framebuffer size
│   ├───Width: [0, 16384]
│   └───Height: [0, 16384]
├───Connectors
│   ├───Connector 0
│   │   ├───Object ID: 49
│   │   ├───Type: eDP
│   │   ├───Status: connected
│   │   ├───Physical size: 260x140 mm
│   │   ├───Subpixel: unknown
│   │   ├───Encoders: {0}
│   │   ├───Modes
│   │   │   ├───1366x768@60.00 preferred driver phsync nvsync
│   │   │   ├───1366x768@40.01 driver phsync nvsync
│   │   │   ├───1280x720@60.00 driver phsync nvsync
│   │   │   ├───1024x768@60.00 driver phsync nvsync
│   │   │   ├───800x600@60.00 driver phsync nvsync
│   │   │   └───640x480@60.00 driver phsync nvsync
│   │   └───Properties
│   │       ├───"EDID" (immutable): blob = 53
│   │       ├───"DPMS": enum {On, Standby, Suspend, Off} = On
│   │       ├───"link-status": enum {Good, Bad} = Good
│   │       ├───"non-desktop" (immutable): range [0, 1] = 0
│   │       ├───"TILE" (immutable): blob = 0
│   │       ├───"CRTC_ID" (atomic): object CRTC = 45
│   │       ├───"scaling mode": enum {None, Full, Center, Full aspect} = None
│   │       ├───"underscan": enum {off, on, auto} = off
│   │       ├───"underscan hborder": range [0, 128] = 0
│   │       ├───"underscan vborder": range [0, 128] = 0
│   │       ├───"max bpc": range [8, 16] = 16
│   │       ├───"HDR_OUTPUT_METADATA": blob = 0
│   │       └───"vrr_capable" (immutable): range [0, 1] = 0
│   ├───Connector 1
│   │   ├───Object ID: 54
│   │   ├───Type: DisplayPort
│   │   ├───Status: disconnected
│   │   ├───Encoders: {1}
│   │   └───Properties
│   │       ├───"EDID" (immutable): blob = 0
│   │       ├───"DPMS": enum {On, Standby, Suspend, Off} = Off
│   │       ├───"link-status": enum {Good, Bad} = Good
│   │       ├───"non-desktop" (immutable): range [0, 1] = 0
│   │       ├───"TILE" (immutable): blob = 0
│   │       ├───"CRTC_ID" (atomic): object CRTC = 0
│   │       ├───"scaling mode": enum {None, Full, Center, Full aspect} = None
│   │       ├───"underscan": enum {off, on, auto} = off
│   │       ├───"underscan hborder": range [0, 128] = 0
│   │       ├───"underscan vborder": range [0, 128] = 0
[... 203 lines cut ...]
    │       ├───"CRTC_Y" (atomic): srange [INT32_MIN, INT32_MAX] = 0
    │       ├───"CRTC_W" (atomic): range [0, INT32_MAX] = 0
    │       ├───"CRTC_H" (atomic): range [0, INT32_MAX] = 0
    │       ├───"SRC_X" (atomic): range [0, UINT32_MAX] = 0
    │       ├───"SRC_Y" (atomic): range [0, UINT32_MAX] = 0
    │       ├───"SRC_W" (atomic): range [0, UINT32_MAX] = 0
    │       └───"SRC_H" (atomic): range [0, UINT32_MAX] = 0
    └───Plane 3
        ├───Object ID: 46
        ├───CRTCs: {1}
        ├───Formats:
        │   └───ARGB8888 (0x34325241)
        └───Properties
            ├───"type" (immutable): enum {Overlay, Primary, Cursor} = Cursor
            ├───"FB_ID" (atomic): object framebuffer = 0
            ├───"IN_FENCE_FD" (atomic): srange [-1, INT32_MAX] = -1
            ├───"CRTC_ID" (atomic): object CRTC = 0
            ├───"CRTC_X" (atomic): srange [INT32_MIN, INT32_MAX] = 0
            ├───"CRTC_Y" (atomic): srange [INT32_MIN, INT32_MAX] = 0
            ├───"CRTC_W" (atomic): range [0, INT32_MAX] = 0
            ├───"CRTC_H" (atomic): range [0, INT32_MAX] = 0
            ├───"SRC_X" (atomic): range [0, UINT32_MAX] = 0
            ├───"SRC_Y" (atomic): range [0, UINT32_MAX] = 0
            ├───"SRC_W" (atomic): range [0, UINT32_MAX] = 0
            └───"SRC_H" (atomic): range [0, UINT32_MAX] = 0`

closed 2026-08-17 by Saviq