canonical/ubuntu-frame #160: Touchscreen and VNC not working after certain events

jrmcpeek, 2023-12-01:
We are running Ubuntu Core 22 on a device with a touchscreen. Currently, we've got the following snap environment:

```
core22            20230801                864    latest/stable  canonical✓  base
mesa-core22       23.0.4                  231    latest/stable  canonical✓  -
network-manager   1.36.6-8                873    22/stable      canonical✓  -
pc-kernel         5.15.0-89.99.2          1517   22/stable      canonical✓  kernel
snapd             2.60.4                  20290  latest/stable  canonical✓  snapd
ubuntu-frame      118-mir2.15.0           7025   22/stable      canonical✓  -
ubuntu-frame-osk  49-squeekboard-v1.17.1  372    22/stable      canonical✓  -
ubuntu-frame-vnc  45-wayvncv0.6.2         226    22/stable      canonical✓  -
```

We're receiving a report from the folks who have the device at their location that shut the device off at night, and then when they turn it on some ~16 hours later, everything boots up like you'd expect, but the touchscreen doesn't work.

While trying to assist them today, this is what I encountered:

* They had forgotten to shut the machine down, so it actually been running overnight.
* It appears the touchscreen device is removed at around 6pm. Monitor being turned off?
* It appears a restart event occurred for ubuntu-frame around 5am.
* After the restart event, ubuntu-frame, ubuntu-frame-osk, ubuntu-frame-vnc, and our onboard kiosk software all appeared to be running as expected. The touchscreen device is NOT restored / re-activated here.
* The user could interact with our software (QR-code based login), which I could see them do via video conference
* After the software logged them in, they were unable to interact with any of the buttons on the screen.

I attempted to initiate a VNC session, which would connected but only present me with a blank screen. A few times it transitioned to a greyscale background, but mostly blank.

Because that did not work, I logged in to the device via SSH. There were no faults logged from our onboard software.

I had the following logged messages from ubuntu-frame-vnc:

```
Dec 01 05:47:09 machine-g6fn30100eem ubuntu-frame-vnc.daemon[507507]: error marshalling arguments for capture_output (signature nio): null value passed for arg 2
Dec 01 05:47:09 machine-g6fn30100eem ubuntu-frame-vnc.daemon[507507]: Error marshalling request: Invalid argument
```

I performed a `snap restart ubuntu-frame-vnc`.

After performing this, I attempted to make a VNC session again, which connected and displayed the screen.

Over VNC, I was able to interact with the buttons in our software, to confirm that our software was not just frozen / ignoring inputs.

Once I confirmed this, I performed a `snap restart ubuntu-frame`.

The display server restarted and our onboard software came back up. The user logged in again, at which point, the touchscreen functionality worked as expected.

Per their claims, this happens "every morning when they turn it on". Obviously in this specific case, it was not a matter of a cold boot event, but perhaps driven from the snap refresh of ubuntu-frame itself?

In all cases, they are able to workaround this issue by pulling the plug on the power cable and plugging it back in.

I have attached the logs related to ubuntu-frame-* here. The touchscreen device is `ILITEK-2911`.

[ubuntu-frame-log.txt](https://github.com/MirServer/ubuntu-frame/files/13528098/ubuntu-frame-log.txt)

Any thoughts?

jrmcpeek, 2023-12-01:
I will check on 22/candidate and let you know.

I'm trying to reproduce locally with 22/beta and if I turn off the monitor, wait 30 seconds, and turn it back on, the display doesn't come back at all.

> What client software were you using?

https://guacamole.apache.org/

jrmcpeek, 2023-12-01:
All right. I'm stuck because I cannot get the display to present itself after power cycling the monitor.

It is the same monitor between both environments, both connected via a USB-C to DisplayPort cable.

It's not clear why the remote one cycles properly but the local one does not recover.

Based on the previous go around with a different bug, I've set debug logging via:

```
snap set ubuntu-frame config="env-hacks=WAYLAND_DEBUG=server"
```

With the following snaps:

```
core22            20230801                864    latest/stable  canonical✓  base
mesa-core22       23.0.4                  231    latest/stable  canonical✓  -
network-manager   1.36.6-8                873    22/stable      canonical✓  -
pc-kernel         5.15.0-89.99.2          1517   22/stable      canonical✓  kernel
snapd             2.60.4                  20290  latest/stable  canonical✓  snapd
ubuntu-frame      118-mir2.15.0           7025   22/stable      canonical✓  -
ubuntu-frame-osk  49-squeekboard-v1.17.1  372    22/stable      canonical✓  -
ubuntu-frame-vnc  45-wayvncv0.6.2         226    22/stable      canonical✓  -
```
[ubuntu-frame-stable-log.txt](https://github.com/MirServer/ubuntu-frame/files/13532381/ubuntu-frame-stable-log.txt)

Key events:
* Boot sequence ends, steady state at 22:05:05 / 4125911.009
* Screen was touched at 22:05:31.234820
* Power button was pressed on the monitor (to turn off) at 22:05:42.910033
* Power button was pressed on the monitor (to turn on) at 22:06:06.575223

Result:
* The screen powered on, showed device branding logos, and then went into standby.
* The screen did not register touch events or attempt to load the device via evdev.

With the following snaps:

```
core22            20230801                864    latest/stable  canonical✓  base
mesa-core22       23.0.4                  231    latest/stable  canonical✓  -
network-manager   1.36.6-8                873    22/stable      canonical✓  -
pc-kernel         5.15.0-89.99.2          1517   22/stable      canonical✓  kernel
snapd             2.60.4                  20290  latest/stable  canonical✓  snapd
ubuntu-frame      123-mir2.16.0           7931   22/candidate   canonical✓  -
ubuntu-frame-osk  49-squeekboard-v1.17.1  372    22/stable      canonical✓  -
ubuntu-frame-vnc  45-wayvncv0.6.2         226    22/stable      canonical✓  -
```

[ubuntu-frame-candidate-log.txt](https://github.com/MirServer/ubuntu-frame/files/13532497/ubuntu-frame-candidate-log.txt)

Key events:
* Boot sequence ends, steady state at 22:34:54 / 1620587.800
* Screen was touched at 22:40:43.943932
* Power button was pressed on the monitor (to turn off) at 22:41:02.922286
* Power button was pressed on the monitor (to turn on) at 22:41:54.382904

Result:
* The screen powered on, showed device branding logos, and then went into standby.
* The screen did not register touch events or attempt to load the device via evdev.

Saviq, 2023-12-02:
Hi @jrmcpeek, with the monitor problems, can you check the output of `drm_info` (or [`graphics-test-tools.drm-info`](https://snapcraft.io/graphics-test-tools/) when the monitor is working, off and back on, not working?

jrmcpeek, 2023-12-04:
Note: Since it sounds like the touchscreen issue I mentioned may already be addressed in `22/candidate`, I've renamed this issue to focus on the screen failing to recover after a power cycle.

Once we get to the bottom of this, I should be able to confirm whether the touchscreen scenario is resolved.

-----

For a different look, I loaded a Classic Live USB based on Ubuntu 22.04.3.

This was used on the same hardware and configuration as the captures running Ubuntu Core.

The display worked as expected when turning it off and back on again.

Here are the logs:
* [classic-startup.txt](https://github.com/MirServer/ubuntu-frame/files/13542656/classic-startup.txt): After power cycling the device and letting it boot to a stable, steady state.
* [classic-off.txt](https://github.com/MirServer/ubuntu-frame/files/13542658/classic-off.txt): After pressing the power button (to turn off) on the monitor and waiting a few seconds.
* [classic-on.txt](https://github.com/MirServer/ubuntu-frame/files/13542665/classic-on.txt): After pressing the power button (to turn on) on the monitor and waiting a few seconds.

And the differences between the files.

```patch
--- classic-startup.txt
+++ classic-off.txt
@@ -395,8 +395,8 @@
     │   │   └───XVYU16161616 (0x38345658)
     │   └───Properties
     │       ├───"type" (immutable): enum {Overlay, Primary, Cursor} = Primary
-    │       ├───"FB_ID" (atomic): object framebuffer = 286
-    │       │   ├───Object ID: 286
+    │       ├───"FB_ID" (atomic): object framebuffer = 288
+    │       │   ├───Object ID: 288
     │       │   ├───Size: 1920x1080
     │       │   ├───Pitch: 7680
     │       │   ├───Bits per pixel: 32
```

```patch
--- classic-off.txt
+++ classic-on.txt
@@ -310,15 +310,15 @@
 │   │   ├───Mode: 1920x1080@60.00 phsync pvsync 
 │   │   └───Properties
 │   │       ├───"ACTIVE" (atomic): range [0, 1] = 1
-│   │       ├───"MODE_ID" (atomic): blob = 291
+│   │       ├───"MODE_ID" (atomic): blob = 290
 │   │       │   └───1920x1080@60.00 phsync pvsync 
 │   │       ├───"OUT_FENCE_PTR" (atomic): range [0, UINT64_MAX] = 0
 │   │       ├───"VRR_ENABLED": range [0, 1] = 0
 │   │       ├───"SCALING_FILTER": enum {Default, Nearest Neighbor} = Default
 │   │       ├───"DEGAMMA_LUT": blob = 0
 │   │       ├───"DEGAMMA_LUT_SIZE" (immutable): range [0, UINT32_MAX] = 128
-│   │       ├───"CTM": blob = 292
-│   │       ├───"GAMMA_LUT": blob = 290
+│   │       ├───"CTM": blob = 286
+│   │       ├───"GAMMA_LUT": blob = 291
 │   │       └───"GAMMA_LUT_SIZE" (immutable): range [0, UINT32_MAX] = 1024
 │   ├───CRTC 1
 │   │   ├───Object ID: 131
@@ -395,8 +395,8 @@
     │   │   └───XVYU16161616 (0x38345658)
     │   └───Properties
     │       ├───"type" (immutable): enum {Overlay, Primary, Cursor} = Primary
-    │       ├───"FB_ID" (atomic): object framebuffer = 288
-    │       │   ├───Object ID: 288
+    │       ├───"FB_ID" (atomic): object framebuffer = 292
+    │       │   ├───Object ID: 292
     │       │   ├───Size: 1920x1080
     │       │   ├───Pitch: 7680
     │       │   ├───Bits per pixel: 32
```

```patch
--- classic-startup.txt
+++ classic-on.txt
@@ -310,15 +310,15 @@
 │   │   ├───Mode: 1920x1080@60.00 phsync pvsync 
 │   │   └───Properties
 │   │       ├───"ACTIVE" (atomic): range [0, 1] = 1
-│   │       ├───"MODE_ID" (atomic): blob = 291
+│   │       ├───"MODE_ID" (atomic): blob = 290
 │   │       │   └───1920x1080@60.00 phsync pvsync 
 │   │       ├───"OUT_FENCE_PTR" (atomic): range [0, UINT64_MAX] = 0
 │   │       ├───"VRR_ENABLED": range [0, 1] = 0
 │   │       ├───"SCALING_FILTER": enum {Default, Nearest Neighbor} = Default
 │   │       ├───"DEGAMMA_LUT": blob = 0
 │   │       ├───"DEGAMMA_LUT_SIZE" (immutable): range [0, UINT32_MAX] = 128
-│   │       ├───"CTM": blob = 292
-│   │       ├───"GAMMA_LUT": blob = 290
+│   │       ├───"CTM": blob = 286
+│   │       ├───"GAMMA_LUT": blob = 291
 │   │       └───"GAMMA_LUT_SIZE" (immutable): range [0, UINT32_MAX] = 1024
 │   ├───CRTC 1
 │   │   ├───Object ID: 131
@@ -395,8 +395,8 @@
     │   │   └───XVYU16161616 (0x38345658)
     │   └───Properties
     │       ├───"type" (immutable): enum {Overlay, Primary, Cursor} = Primary
-    │       ├───"FB_ID" (atomic): object framebuffer = 286
-    │       │   ├───Object ID: 286
+    │       ├───"FB_ID" (atomic): object framebuffer = 292
+    │       │   ├───Object ID: 292
     │       │   ├───Size: 1920x1080
     │       │   ├───Pitch: 7680
     │       │   ├───Bits per pixel: 32
```

jrmcpeek, 2023-12-04:
@Saviq 

> Is the kernel the same?…

Ubuntu Core:

```
pc-kernel            5.15.0-89.99.2          1517   22/stable      canonical✓  kernel
Linux machine-1234567890 5.15.0-89-generic #99-Ubuntu SMP Mon Oct 30 20:42:41 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

Live USB:

```
Linux ubuntu 6.2.0-26 generic #26-22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

AlanGriffiths, 2023-12-04:
> Is there a target timeline for the promotion of `22/candidate` to `22/stable` for the mir2.16 updates?

Here's the announcement: https://forum.snapcraft.io/t/call-for-testing-ubuntu-frame-mir-kiosk/37963

We'll announce the promotion there. (Given the holiday season, "we aim to promote to stable in two weeks" may well become early January)