canonical/ubuntu-frame #162: can we add the title-bar to the application?

timiil, 2024-01-10:
hello, as the title say, in some env , we may need to give the user press the 'X' button to exit the application; because in emmbed device may not have keyboard, so the user can not press ctrl+F4, ctrl +Q . and the application like 'gedit', 'firefox', when they show in ubuntu-frame mode, they dont have the title-bar at all, so , how do we can make the user to exit the app?

Saviq, 2024-01-10:
Hi @timiil, all applications in Frame are fullscreen, it's on the application to have the necessary controls.

Why would you need to quit applications? It doesn't obviously fit the use case Frame is covering. How would you then open the closed application again?

timiil, 2024-01-10:
let's say we start an app "my apps launcher" first in Ubuntu-frame， which have serval buttons :["gedit","firefox","rythmbox"]，user can press the button and saw the app in modal mode，user can only interact the current app which open by him，till he turn off (close) the app，he can press the launcher's buttons again.

Saviq, 2024-01-10:
@timiil the launcher should then have means to manage the applications it launched. Through some privileged Wayland protocols (https://wayland.app/protocols/wlr-foreign-toplevel-management-unstable-v1, https://wayland.app/protocols/wlr-layer-shell-unstable-v1) you can build a launcher that's always on screen and allows application management.

I theory we could have a mode in Frame to keep applications maximized, rather than fullscreen, which would cause them to draw their titlebars.

Have a look at https://github.com/MirServer/ubuntu-frame/blob/main/src/frame_window_manager.cpp and see if you can modify it to work the way you want. If you can propose this change, we'll consider it for inclusion into Frame itself.

timiil, 2024-01-10:
thanks for your reply. i will take a look for the code you listed above.
 
let's make my request clear again pls: 

we should let the root app(my apps launcher) run in frame by full screen mode. 

but all children gui apps those whom created by the root app should all run in "maximize mode" in Ubuntu-frame，so the use have the close button to click.

Saviq, 2024-01-10:
@timiil how would Frame know which is the launcher app, and which are "normal" apps?

You need to make your launcher special, and the above mentioned protocol extensions allow that - you can make your launcher fullscreen, and sit below other applications.

timiil, 2024-01-10:
thanks for your reply. i will take a look for the code you listed above.
 
let's make my request clear again pls: 

we should let the root app(my apps launcher) run in frame by full screen mode. 

but all children gui apps those whom created by the root app should all run in "maximize mode" in Ubuntu-frame，so the use have the close button to click.

> @timiil how would Frame know which is the launcher app, and which are "normal" apps?
> 
> You need to make your launcher special, and the above mentioned protocol extensions allow that - you can make your launcher fullscreen, and sit below other applications.

i am just iamge something like openbox that we could config every window one by one...

but your solution also seems  ok，let's make the launcher borderless or notcloseable or make some scripts deamon on it.

timiil, 2024-01-11:
as your guideline, i am trying these:

<a>https://github.com/MirServer/ubuntu-frame/blob/main/src/frame_window_manager.cpp</a>

```
void FrameWindowManagerPolicy::apply_bespoke_fullscreen_placement(
    WindowSpecification& specification, WindowInfo const& window_info) const
{
    if (maximize_mode) {
        specification.state() = mir_window_state_maximized;
        // Omitting the call to tools.place_and_size_for_state and fullscreen state setting
    } else {
        // Original implementation
        specification.state() = mir_window_state_fullscreen;
        tools.place_and_size_for_state(specification, window_info);
        specification.state() = mir_window_state_fullscreen;
    }
}
```

as if we add the new command line option maximize_mode, seems we need to modify the MirRunner and DisplayConfiguration  which they are NOT inside this project ?

Saviq, 2024-01-11:
@timiil Frame-specific options are defined here, no need to modify MirRunner:

https://github.com/MirServer/ubuntu-frame/blob/b1504c6bebe872313458e6bbcaea296cd8b765a4/src/frame_main.cpp#L48-L72

timiil, 2024-01-11:
like this ?
```
bool maximize_mode = false; // Global flag

int main(int argc, char const* argv[])
{
    using namespace miral;
    MirRunner runner{argc, argv};

    // ... existing code ...

    runner.run_with({
        // ... other configuration options ...

        ConfigurationOption{
            [&](bool option) { maximize_mode = option; },
            "maximize-mode", "Enable maximize mode for windows", false
        },

        // ... rest of the configuration options ...
    });

    // ...
}
```

Saviq, 2024-01-11:
Yes, you just need an extra `ConfigurationOption` there.

AlanGriffiths, 2024-01-11:
Rather than hacking Frame, why not use mir-kiosk? That already 

1. maximises regular windows by default
2. leaves fullscreen windows as fullscreen

Which is what I think the OP is asking for