canonical/ubuntu-frame #163: Screen rotated 180 degrees after 123-mir2.16.1 update

freefos, 2024-01-11:
Hi,

After the 123-mir2.16.1 update, the frame screen is rotated 180 degrees.

[image]
The boot animation is still correct.

Reverted back to 121-mir2.15.0 and everything is back to normal.

This is my display configuration

```
layouts:
# keys here are layout labels (used for atomically switching between them).
# The yaml anchor 'the_default' is used to alias the 'default' label

  default:
    cards:
    # a list of cards (currently matched by card-id)

    - card-id: 0
      HDMI-A-1:
        # This output supports the following modes: 1920x1080@60.0, 1920x1080@59.9,
        # 1920x1080@30.0, 1920x1080@30.0, 1920x1080@50.0, 1920x1080@25.0, 1600x900@60.0,
        # 1280x1024@75.0, 1280x1024@60.0, 1152x864@75.0, 1280x720@60.0, 1280x720@59.9,
        # 1280x720@50.0, 1024x768@75.0, 1024x768@60.0, 800x600@75.0, 800x600@60.3,
        # 720x576@50.0, 720x480@60.0, 720x480@59.9, 640x480@75.0, 640x480@60.0,
        # 640x480@59.9, 720x400@70.1
        #
        # Uncomment the following to enforce the selected configuration.
        # Or amend as desired.
        #
        state: enabled  # {enabled, disabled}, defaults to enabled
        mode: 1920x1080@60.0    # Defaults to preferred mode
        position: [0, 0]        # Defaults to [0, 0]
        orientation: left       # {normal, left, right, inverted}, defaults to normal
        scale: 1
        group: 0        # Outputs with the same non-zero value are treated as a single display

  side_by_side:
    cards:
    # a list of cards (currently matched by card-id)

    - card-id: 0
      HDMI-A-1:
        # This output supports the following modes: 1920x1080@60.0, 1920x1080@59.9,
        # 1920x1080@30.0, 1920x1080@30.0, 1920x1080@50.0, 1920x1080@25.0, 1600x900@60.0,
        # 1280x1024@75.0, 1280x1024@60.0, 1152x864@75.0, 1280x720@60.0, 1280x720@59.9,
        # 1280x720@50.0, 1024x768@75.0, 1024x768@60.0, 800x600@75.0, 800x600@60.3,
        # 720x576@50.0, 720x480@60.0, 720x480@59.9, 640x480@75.0, 640x480@60.0,
        # 640x480@59.9, 720x400@70.1
        #
        # Uncomment the following to enforce the selected configuration.
        # Or amend as desired.
        #
        state: enabled  # {enabled, disabled}, defaults to enabled
        mode: 1920x1080@60.0    # Defaults to preferred mode
        position: [0, 0]        # Defaults to [0, 0]
        orientation: left       # {normal, left, right, inverted}, defaults to normal
        scale: 1
        group: 0        # Outputs with the same non-zero value are treated as a single display
```

What could be the issue here?

Saviq, 2024-01-11:
Hi @freefos, sorry that happened to you… can you post frame's logs for the initial display configuration?

I suspect the output name might have changed, and your previous configuration misses the setting for the new output name…

I will also suggest you subscribe to https://discourse.ubuntu.com/c/mir/15, we had this version in the `candidate` channel for a few weeks, would've been good if you tested it before.

Saviq, 2024-01-11:
@freefos I just confirmed that this behaves correctly on all the hardware I have to hand…

```plain
$ snap install ubuntu-frame --revision 7257
ubuntu-frame 121-mir2.15.0 from Canonical✓ installed

# … Frame displays in "normal" orientation

$ snap get ubuntu-frame display
```
```yaml
layouts:
# keys here are layout labels (used for atomically switching between them).                                                                                                                               
# The yaml anchor 'the_default' is used to alias the 'default' label                                                                                                                                      
                                                  
  default:
    cards:
    # a list of cards (currently matched by card-id)
                                                  
    - card-id: 0          
      HDMI-A-1:
        # This output supports the following modes: 1920x1200@60.0, 1920x1080@60.0,
        # 1920x1080@59.9, 1920x1080@30.0, 1920x1080@30.0, 1920x1080@50.0, 1920x1080@25.0,
        # 1920x1080@24.0, 1920x1080@24.0, 1600x1200@60.0, 1280x1024@75.0, 1280x1024@60.0,
        # 1152x864@75.0, 1280x720@60.0, 1280x720@59.9, 1280x720@50.0, 1024x768@75.0,
        # 1024x768@60.0, 800x600@75.0, 800x600@60.3, 720x576@50.0, 720x576@25.0,
        # 720x480@60.0, 720x480@59.9, 720x480@30.0, 720x480@30.0, 640x480@75.0,
        # 640x480@60.0, 640x480@59.9, 720x400@70.1 
        #
        # Uncomment the following to enforce the selected configuration.
        # Or amend as desired.                                                                       
        #                                                                                                                                                                                                 
        state: enabled  # {enabled, disabled}, defaults to enabled
        mode: 1920x1200@60.0    # Defaults to preferred mode
        position: [0, 0]        # Defaults to [0, 0]
        orientation: normal     # {normal, left, right, inverted}, defaults to normal
        scale: 1
        group: 0        # Outputs with the same non-zero value are treated as a single display
                                                  
  side_by_side:                                                                                                                                                                                           
    cards:
    # a list of cards (currently matched by card-id)

    - card-id: 0
      HDMI-A-1:
        # This output supports the following modes: 1920x1200@60.0, 1920x1080@60.0,
        # 1920x1080@59.9, 1920x1080@30.0, 1920x1080@30.0, 1920x1080@50.0, 1920x1080@25.0,
        # 1920x1080@24.0, 1920x1080@24.0, 1600x1200@60.0, 1280x1024@75.0, 1280x1024@60.0,
        # 1152x864@75.0, 1280x720@60.0, 1280x720@59.9, 1280x720@50.0, 1024x768@75.0,
        # 1024x768@60.0, 800x600@75.0, 800x600@60.3, 720x576@50.0, 720x576@25.0,
        # 720x480@60.0, 720x480@59.9, 720x480@30.0, 720x480@30.0, 640x480@75.0,
        # 640x480@60.0, 640x480@59.9, 720x400@70.1 
        #
        # Uncomment the following to enforce the selected configuration.
        # Or amend as desired.
        #
        state: enabled  # {enabled, disabled}, defaults to enabled
        mode: 1920x1200@60.0    # Defaults to preferred mode
        position: [0, 0]        # Defaults to [0, 0]
        orientation: normal     # {normal, left, right, inverted}, defaults to normal
        scale: 1
        group: 0        # Outputs with the same non-zero value are treated as a single display
```

```plain
$ snap set ubuntu-frame display="                                                       
layouts:
  default:
    cards:
    - card-id: 0
      HDMI-A-1:
        orientation: left
"

# Frame displays in `left` orientation
```

```plain
$ snap refresh ubuntu-frame --channel 22/stable
ubuntu-frame (22/stable) 123-mir2.16.1 from Canonical✓ refreshed

# Frame displays in `left` orientation
```

I would also suggest trimming your `display` configuration to just the values you want changed, no need for the comments and defaults, they make it trickier to read.

Saviq, 2024-01-12:
Hi @freefos, any update on this? Were you able to solve this?