canonical/ubuntu-frame #165: When run as a user (not daemon) the cursor is not visible

AlanGriffiths, 2024-01-29:
Doesn't happen on Intel, doesn't happen when run as a daemon but seen on RPi400.

1. Log into a VT on a classic system
2. As the same user (on that VT or via ssh) run `ubuntu-frame`

**Expect:** cursor moves with mouse
**Actual:** cursor not visible

Note: this is probably confinement related as refreshing with `--devmode` gives the expected behaviour

AlanGriffiths, 2024-01-29:
Further note: I don't see this with `mir-kiosk`

mb-shorai, 2024-03-07:
Jeez, I've spent quite some time thinking something is wrong with my setup. I can reproduce it in VirtualBox with Ubuntu 22.04 when running `ubuntu-frame` as a daemon vs user. Also, it works with `--devmode` indeed.

There are a couple of deny actions in the journal:
```
Mar 07 15:44:15 max audit[1045]: USER_AVC pid=1045 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/login1/seat/seat0" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.7" pid=5166 label="snap.ubuntu-frame.ubuntu-frame" peer_pid=1057 peer_label="unconfined"
                                  exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
Mar 07 15:44:15 max audit[5166]: AVC apparmor="DENIED" operation="open" profile="snap.ubuntu-frame.ubuntu-frame" name="/run/user/1000/#104" pid=5166 comm="frame" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
Mar 07 15:44:15 max audit[5166]: AVC apparmor="DENIED" operation="open" profile="snap.ubuntu-frame.ubuntu-frame" name="/dev/shm/#11" pid=5166 comm="frame" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
```