canonical/ubuntu-frame #171: malloc_consolidate(): unaligned fastbin chunk detected

ryangrunest, 2024-03-08:
ubuntu-frame is freezing at seemingly random points. 

```
 % sudo snap logs ubuntu-frame
2024-03-06T22:56:04Z ubuntu-frame.daemon[394759]: [2024-03-06 22:56:04.004627] <information> evdev-input: Added /dev/input/event8: Touch KB9000
2024-03-06T22:56:04Z ubuntu-frame.daemon[394759]: [2024-03-06 22:56:04.011304] <information> evdev-input: Added /dev/input/event10: Touch KB9000
2024-03-06T22:56:04Z ubuntu-frame.daemon[394759]: [2024-03-06 22:56:04.017292] <information> evdev-input: Added /dev/input/event9: Touch KB9000
2024-03-06T22:56:04Z ubuntu-frame.daemon[394759]: [2024-03-06 22:56:04.021692] <information> evdev-input: Added /dev/input/event4: RPI Wired Keyboard 4
2024-03-06T22:56:04Z ubuntu-frame.daemon[394759]: [2024-03-06 22:56:04.026211] <information> evdev-input: Added /dev/input/event5: RPI Wired Keyboard 4
2024-03-06T22:56:04Z ubuntu-frame.daemon[394759]: [2024-03-06 22:56:04.030774] <information> evdev-input: Added /dev/input/event6
2024-03-06T22:56:04Z ubuntu-frame.daemon[394759]: [2024-03-06 22:56:04.035781] <information> evdev-input: Added /dev/input/event11
2024-03-06T22:56:04Z ubuntu-frame.daemon[394759]: [2024-03-06 22:56:04.041764] <information> evdev-input: Added /dev/input/event12
2024-03-07T23:01:52Z ubuntu-frame.daemon[394759]: malloc_consolidate(): unaligned fastbin chunk detected
2024-03-07T23:01:52Z ubuntu-frame.daemon[394759]: !!! Fatal signal received. Attempting cleanup, but deadlock may occur 
```

```
 % free -m 
               total        used        free      shared  buff/cache   available
Mem:           63692        2715       55441         134        5534       60130
Swap:           8191           0        8191 
```

CPU usage was also below 15%. 

What can I do to further diagnose and resolve this issue?

Saviq, 2024-03-08:
Hi @ryangrunest, thanks for this report, can you please confirm which version of Frame you're running?

```
$ snap info ubuntu-frame | grep -e installed -e tracking
```

Also, can you provide more details about your setup? Looks like you're running on a Raspberry Pi?

Saviq, 2024-03-11:
Hi again. It's the first time we've seen this, and that's very little information to go on… Neither Mir or Frame themselves call `malloc_consolidate`, so it would have to be in a dependency or a driver, maybe…

If you have any means to run Frame [under gdb](https://snapcraft.io/docs/using-gdb-gdbserver), maybe that could point at a culprit… https://ubuntu.com/server/docs/service-debuginfod can provide more symbols.

How often does it happen for you?

ryangrunest, 2024-03-12:
I'm hoping to get a chance to gather gdb data tomorrow. I will update the thread as soon as I have it. Thank you

robinelvin, 2024-06-18:
I am experiencing the same issue. This is on a PC with dual monitor display. Seems to be hanging every 24 hours or so

```
2024-06-18T06:22:34Z ubuntu-frame.daemon[1079]: malloc_consolidate(): unaligned fastbin chunk detected
2024-06-18T06:22:34Z ubuntu-frame.daemon[1079]: !!! Fatal signal received. Attempting cleanup, but deadlock may occur
```

```
snap info ubuntu-frame | grep -e installed -e tracking
tracking:     22/stable
installed:          144-mir2.17.0                    (9750) 6MB -
```

Saviq, 2024-06-18:
Can you all please let us know what graphics hardware you're on?

Also migrating to the `24` track in case that got fixed in e.g. the GPU drivers?

```
snap refresh ubuntu-frame --channel 24/stable
```

Thanks!

robinelvin, 2024-06-18:
@Saviq Hardware is `AMD Ryzen Embedded R1102G with Radeon Vega Gfx`

Switching to 24/stable and will report back - thanks

robinelvin, 2024-07-09:
@Saviq Seems to be stable on the `24` track - thank you

Saviq, 2024-07-15:
Since both reports of this were on AMD graphics, and using the newer Mesa release helps - I'm going to close this pending any subsequent reports.

closed 2026-08-17 by Saviq