canonical/ubuntu-frame #176: Electron app not starting

J-Cake, 2024-04-13:
I would give more info, but I simply don't have any...

I want to run Obsidian on Frame but am struggling to get very far. Ubuntu Frame starts and shows the typical greydient (lool) but when running Obsidian, both via the CLI directly and via `frame-it`, I'm unsuccessful. Below is the output from `WAYLAND-DISPLAY=wayland-99 sudo -E obsidian --enable-features=UseOzonePlatform --ozone-platform=wayland`, but I've tried all sorts of combinations, read through basically all forums etc. Hoping for some help.

Thanks

```
jcake@jcake-edu:~$ sudo frame-it obsidian --enable-features=UseOzonePlatform --ozone-platform=wayland 
+ [ /snap/ubuntu-frame/8587/bin/run-frame = --help ]
+ mkdir -p /root/snap/ubuntu-frame/8587/.config/
+ grep -vE (^vt=|^console-provider=vt) /var/snap/ubuntu-frame/8587/frame.config
+ rm -f /root/snap/ubuntu-frame/8587/.config/frame.display
+ cp /var/snap/ubuntu-frame/8587/frame.display /root/snap/ubuntu-frame/8587/.config/
cp: cannot stat '/var/snap/ubuntu-frame/8587/frame.display': No such file or directory
+ true
+ exec /snap/ubuntu-frame/8587/bin/run-frame /snap/ubuntu-frame/8587/bin/graphics-core22-wrapper /snap/ubuntu-frame/8587/usr/local/bin/frame
+ [ /snap/ubuntu-frame/8587/bin/graphics-core22-wrapper = --help ]
+ dirname /run/user/0/snap.ubuntu-frame
+ XDG_RUNTIME_DIR=/run/user/0
+ export XDG_RUNTIME_DIR
+ mkdir -p /run/user/0 -m 700
+ [ -n wayland-0 ]
+ [ -e /run/user/0/wayland-0 ]
+ exec /snap/ubuntu-frame/8587/bin/graphics-core22-wrapper /snap/ubuntu-frame/8587/usr/local/bin/frame
[2024-04-13 21:30:54.157165] <information> mirserver: Starting
[2024-04-13 21:30:54.160464] < - debug - > mirserver: Not using logind for session management: Failed to find active session
[2024-04-13 21:30:54.160654] < - debug - > mirserver: Using Linux VT subsystem for session management
[2024-04-13 21:30:54.160740] <information> mircommon: Loading modules from: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform
[2024-04-13 21:30:54.160766] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/graphics-eglstream-kms.so.21
[2024-04-13 21:30:54.160776] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.21
[2024-04-13 21:30:54.160781] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/renderer-egl-generic.so.21
[2024-04-13 21:30:54.160786] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/server-x11.so.21
[2024-04-13 21:30:54.160790] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/input-evdev.so.8
[2024-04-13 21:30:54.161175] <information> mirserver: Found display driver: mir:eglstream-kms (version 2.16.3)
[2024-04-13 21:30:54.161663] < - debug - > eglstream: Found EGLDeviceEXT with device extensions: EGL_MESA_device_software EGL_EXT_device_drm_render_node
[2024-04-13 21:30:54.161689] <information> eglstream: Failed to query DRM node for EGLDevice: Failed to determine DRM device node path from EGLDevice: EGL_BAD_PARAMETER (0x300c)
[2024-04-13 21:30:54.161693] < - debug - > eglstream: EGLDeviceEXTs found, but none are suitable for Mir
[2024-04-13 21:30:54.161697] <information> mirserver: (Unsupported by system environment)
[2024-04-13 21:30:54.161702] <information> mirserver: Found display driver: mir:gbm-kms (version 2.16.3)
[2024-04-13 21:30:54.161867] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card0, driver bochs-drm
pci id for fd 12: 1234:1111, driver (null)
MESA-LOADER: failed to open bochs-drm: /snap/ubuntu-frame/8587/graphics/usr/lib/i386-linux-gnu/dri//bochs-drm_dri.so: cannot open shared object file: No such file or directory (search paths /snap/ubuntu-frame/8587/graphics/usr/lib/x86_64-linux-gnu/dri/:/snap/ubuntu-frame/8587/graphics/usr/lib/i386-linux-gnu/dri/, suffix _dri)
pci id for fd 11: 1234:1111, driver (null)
kmsro: driver missing
Failed to create /root/.cache for shader cache (Permission denied)---disabling.
[2024-04-13 21:30:54.175964] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver 
[2024-04-13 21:30:54.176044] <information> mirserver: Driver supports:
[2024-04-13 21:30:54.176053] <information> mirserver:   /devices/pci0000:00/0000:00:02.0/drm/card0 (priority 256)
[2024-04-13 21:30:54.176062] <information> mirserver: Found display driver: mir:egl-generic (version 2.16.3)
[2024-04-13 21:30:54.176087] <information> mirserver: Found display driver: mir:x11 (version 2.16.3)
[2024-04-13 21:30:54.176101] <information> mirserver: (Unsupported by system environment)
[2024-04-13 21:30:54.176117] <information> mirserver: Selected display driver: mir:gbm-kms (version 2.16.3) for device ((null): /dev/dri/card0)
pci id for fd 12: 1234:1111, driver (null)
MESA-LOADER: failed to open bochs-drm: /snap/ubuntu-frame/8587/graphics/usr/lib/i386-linux-gnu/dri//bochs-drm_dri.so: cannot open shared object file: No such file or directory (search paths /snap/ubuntu-frame/8587/graphics/usr/lib/x86_64-linux-gnu/dri/:/snap/ubuntu-frame/8587/graphics/usr/lib/i386-linux-gnu/dri/, suffix _dri)
pci id for fd 11: 1234:1111, driver (null)
kmsro: driver missing
[2024-04-13 21:30:54.178632] <information> mircommon: Loading modules from: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform
[2024-04-13 21:30:54.178657] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/graphics-eglstream-kms.so.21
[2024-04-13 21:30:54.178664] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.21
[2024-04-13 21:30:54.178669] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/renderer-egl-generic.so.21
[2024-04-13 21:30:54.178674] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/server-x11.so.21
[2024-04-13 21:30:54.178678] <information> mircommon: Loading module: /snap/ubuntu-frame/8587/usr/lib/x86_64-linux-gnu/mir/server-platform/input-evdev.so.8
[2024-04-13 21:30:54.179053] <information> mirserver: Found rendering driver: mir:eglstream-kms (version 2.16.3)
[2024-04-13 21:30:54.179184] < - debug - > eglstream: Failed to find kernel device for EGLDevice: Failed to determine DRM device node path from EGLDevice: EGL_BAD_PARAMETER (0x300c)
[2024-04-13 21:30:54.179189] < - debug - > eglstream: EGLDeviceEXTs found, but none are suitable for Mir
[2024-04-13 21:30:54.179192] <information> mirserver: (Unsupported by system environment)
[2024-04-13 21:30:54.179196] <information> mirserver: Found rendering driver: mir:gbm-kms (version 2.16.3)
[2024-04-13 21:30:54.179202] < - debug - > gbm-kms: GBM-capable display found
[2024-04-13 21:30:54.179361] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card0, driver bochs-drm
pci id for fd 13: 1234:1111, driver (null)
[... 114 lines cut ...]
[2024-04-13 21:30:54.206175] <information> GLRenderer: GL version: OpenGL ES 3.2 Mesa 23.0.4-0ubuntu1~22.04.1
[2024-04-13 21:30:54.206179] <information> GLRenderer: GLSL version: OpenGL ES GLSL ES 3.20
[2024-04-13 21:30:54.206209] <information> GLRenderer: GL extensions: GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_filter_anisotropic GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_float_linear GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_NV_conditional_render GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_NV_pack_subimage GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_OES_viewport_array GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compress [... cut ...]
[2024-04-13 21:30:54.208022] < -warning- > miral: Attempt to read custom output attribute (surface-title) that wasn't added
[2024-04-13 21:30:54.208033] < -warning- > miral: Attempt to read custom output attribute (snap-name) that wasn't added
[2024-04-13 21:31:01.160244] <information> evdev-input: Added /dev/input/event0
[2024-04-13 21:31:01.163231] <information> evdev-input: Added /dev/input/event1
[2024-04-13 21:31:01.164826] <information> evdev-input: Added /dev/input/event3
[2024-04-13 21:31:01.164874] <information> evdev-input: Added /dev/input/event2
```

AlanGriffiths, 2024-04-14:
> jcake@jcake-edu:~$ sudo frame-it obsidian --enable-features=UseOzonePlatform --ozone-platform=wayland

Why are you running `frame-it` as root? What happens with `frame-it obsidian --enable-features=UseOzonePlatform --ozone-platform=wayland`?

> I've tried all sorts of combinations, read through basically all forums etc. Hoping for some help

This is probably what you need:

https://mir-server.io/docs/packaging-an-electron-application-as-an-iot-gui

Closing issue as this is not a problem with Ubuntu Frame

closed 2024-04-14 by AlanGriffiths

J-Cake, 2024-04-14:
I'm running `frame-it` as root to make sure it isn't that. As I mentioned, I've tried basically everything. I've indeed been following that article too, and have had very little luck.

AlanGriffiths, 2024-04-15:
> I'm running `frame-it` as root to make sure it isn't that. As I mentioned, I've tried basically everything. I've indeed been following that article too, and have had very little luck.

Running as root _will cause problems_: the log from doing so is unlikely to provide useful information on why Obsidian doesn't do what you expect. 

> Ubuntu Frame starts and shows the typical greydient

This shows that Frame starts and implies that any problem is not with Frame.

> when running Obsidian, both via the CLI directly

So far as I can see Obsidian is not in the Ubuntu archive, so any problem is likely to be with the way it is built, packaged or installed.

Errors from Obsidian (ideally run via the CLI) might indicate why it doesn't start

J-Cake, 2024-04-15:
I see. When I'm at my computer next, I'll run it again as standard. Obsidian is a available as both a snap and directly in the package repos. I've tried both. I also tried it with VSCode packaged in the same (or at least similar) way and had the same results. 

I also forgot to mention that running frame-it on my desktop the way the article describes did work, but only if the electron app is started with the necessary wayland parameters