canonical/ubuntu-frame #183: Ubuntu Frame no longser starting on Raspberry Pi CM4 (Ubuntu Core 22) after latest update (134-mir2.16.4)

jaydensmith, 2024-04-26:
Everything was working fine up until this update. I don't see any errors in journalctl except for:

```
Apr 26 13:39:48 ubuntu audit[1746]: AVC apparmor="DENIED" operation="open" profile="snap.ubuntu-frame.daemon" name="/run/user/0/" pid=1746 comm="mkdir" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
Apr 26 13:39:48 ubuntu kernel: audit: type=1400 audit(1714102788.615:57): apparmor="DENIED" operation="open" profile="snap.ubuntu-frame.daemon" name="/run/user/0/" pid=1746 comm="mkdir" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
```

The full contents of the ubuntu-frame logs are:

```
2024-04-26T13:39:14+10:00 systemd[1]: Starting Service for snap application ubuntu-frame.daemon...
2024-04-26T13:39:24+10:00 systemd[1]: Started Service for snap application ubuntu-frame.daemon.
2024-04-26T13:39:39+10:00 ubuntu-frame.daemon[1609]: ++ snapctl get display
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + '[' -z '' ']'
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1716]: + initialise-display-config
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + disown
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + exec nohup /snap/ubuntu-frame/9125/bin/run-frame /snap/ubuntu-frame/9125/bin/graphics-core22-wrapper /snap/ubuntu-frame/9125/usr/local/bin/frame
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1717]: + sh -c 'sleep 5; mkdir "/run/user/007"; rmdir "/run/user/007"'
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + [ /snap/ubuntu-frame/9125/bin/graphics-core22-wrapper = --help ]
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1745]: + dirname /run/user/0/snap.ubuntu-frame
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + XDG_RUNTIME_DIR=/run/user/0
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + export XDG_RUNTIME_DIR
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + mkdir -p /run/user/0 -m 700
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1718]: Setting up watches.
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1718]: Watches established.
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + [ -n  ]
2024-04-26T13:39:48+10:00 ubuntu-frame.daemon[1501]: + exec /snap/ubuntu-frame/9125/bin/graphics-core22-wrapper /snap/ubuntu-frame/9125/usr/local/bin/frame
2024-04-26T13:39:49+10:00 ubuntu-frame.daemon[1501]: [2024-04-26 13:39:49.454034] <information> mirserver: Starting
```

RAOF, 2024-04-26:
That is very strange!

If you revert to the previous ubuntu-frame revision (`snap revert ubuntu-frame` should do it; you might then wish to `snap refresh --hold ubuntu-frame` to prevent auto-updates) does the old version of ubuntu-frame continue to work?

RAOF, 2024-04-26:
Ok. I've got a Pi4 (but not a CM4); I'll see if I can reproduce here.

RAOF, 2024-04-26:
Hm, no. I can't reproduce this on my Pi4. I've installed the new ubuntu-frame, it's in daemon mode, and rebooted, and it still comes up.

RAOF, 2024-04-26:
Hm. Based upon when your logs terminate, ubuntu-frame *might* be hanging trying to acquire the VT?

jaydensmith, 2024-04-28:
It's definitely hanging doing something. When trying to refresh/remove `snap` hangs waiting for the process to end before it eventually terminates ubuntu-frame. I've just tried with a brand new CM4 and carrier board, same issue unfortunately.

```
2024-04-29T09:41:22+10:00 systemd[1]: Stopping Service for snap application ubuntu-frame.daemon...
2024-04-29T09:41:42+10:00 ubuntu-frame.daemon[1186]: terminate called after throwing an instance of 'boost::wrapexcept<std::runtime_error>'
2024-04-29T09:41:42+10:00 ubuntu-frame.daemon[1186]:   what():  Compositor thread failed to stop
2024-04-29T09:41:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: State 'stop-sigterm' timed out. Killing.
2024-04-29T09:41:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1186 (frame) with signal SIGKILL.
2024-04-29T09:41:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1336 (frame) with signal SIGKILL.
2024-04-29T09:41:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1337 (frame) with signal SIGKILL.
2024-04-29T09:41:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1338 (frame) with signal SIGKILL.
2024-04-29T09:41:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1341 (Mir/Comp) with signal SIGKILL.
2024-04-29T09:41:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1345 (Mir/Workqueue) with signal SIGKILL.
2024-04-29T09:42:22+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Processes still around after SIGKILL. Ignoring.
2024-04-29T09:42:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: State 'final-sigterm' timed out. Killing.
2024-04-29T09:42:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1186 (frame) with signal SIGKILL.
2024-04-29T09:42:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1336 (frame) with signal SIGKILL.
2024-04-29T09:42:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1337 (frame) with signal SIGKILL.
2024-04-29T09:42:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1338 (frame) with signal SIGKILL.
2024-04-29T09:42:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1341 (Mir/Comp) with signal SIGKILL.
2024-04-29T09:42:52+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Killing process 1345 (Mir/Workqueue) with signal SIGKILL.
2024-04-29T09:43:22+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Processes still around after final SIGKILL. Entering failed mode.
2024-04-29T09:43:22+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Failed with result 'timeout'.
2024-04-29T09:43:22+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Unit process 1186 (frame) remains running after unit stopped.
2024-04-29T09:43:22+10:00 systemd[1]: Stopped Service for snap application ubuntu-frame.daemon.
2024-04-29T09:43:22+10:00 systemd[1]: snap.ubuntu-frame.daemon.service: Consumed 10.397s CPU time.
```

@Saviq I do get more logs with 2.16.3, but the issue 

When I install from fresh, it runs:
```
2024-04-29T09:36:06+10:00 systemd[1]: Started Service for snap application ubuntu-frame.daemon.
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3081]: ++ snapctl get display
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + '[' -z '' ']'
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3086]: + initialise-display-config
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + disown
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3087]: + sh -c 'sleep 5; mkdir "/run/user/007"; rmdir "/run/user/007"'
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + exec nohup /snap/ubuntu-frame/8737/bin/run-frame /snap/ubuntu-frame/8737/bin/graphics-core22-wrapper /snap/ubuntu-frame/8737/usr/local/bin/frame
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3088]: Setting up watches.
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3088]: Watches established.
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + [ /snap/ubuntu-frame/8737/bin/graphics-core22-wrapper = --help ]
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3092]: + dirname /run/user/0/snap.ubuntu-frame
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + XDG_RUNTIME_DIR=/run/user/0
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + export XDG_RUNTIME_DIR
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + mkdir -p /run/user/0 -m 700
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + [ -n  ]
2024-04-29T09:36:06+10:00 ubuntu-frame.daemon[3063]: + exec /snap/ubuntu-frame/8737/bin/graphics-core22-wrapper /snap/ubuntu-frame/8737/usr/local/bin/frame
2024-04-29T09:36:07+10:00 ubuntu-frame.daemon[3063]: [2024-04-29 09:36:07.405102] <information> mirserver: Starting
2024-04-29T09:36:07+10:00 ubuntu-frame.daemon[3063]: [2024-04-29 09:36:07.419932] < - debug - > mirserver: Using Linux VT subsystem for session management
2024-04-29T09:36:07+10:00 ubuntu-frame.daemon[3063]: [2024-04-29 09:36:07.423174] <information> mircommon: Loading modules from: /snap/ubuntu-frame/8737/usr/lib/aarch64-linux-gnu/mir/server-platform
2024-04-29T09:36:07+10:00 ubuntu-frame.daemon[3063]: [2024-04-29 09:36:07.423784] <information> mircommon: Loading module: /snap/ubuntu-frame/8737/usr/lib/aarch64-linux-gnu/mir/server-platform/graphics-eglstream-kms.so.21
[... cut ...]
2024-04-29T09:36:09+10:00 ubuntu-frame.daemon[3063]: [2024-04-29 09:36:09.985350] <information> mirserver: . |_ Scaling factor: 1.00
2024-04-29T09:36:09+10:00 ubuntu-frame.daemon[3063]: [2024-04-29 09:36:09.985472] < -warning- > miral: Attempt to read custom output attribute (surface-title) that wasn't added
2024-04-29T09:36:09+10:00 ubuntu-frame.daemon[3063]: [2024-04-29 09:36:09.985583] < -warning- > miral: Attempt to read custom output attribute (snap-name) that wasn't added
```

When rebooting the system, the wallpaper never appears on the screen, just tty.

jaydensmith, 2024-04-29:
Interestingly, I have a device that has not refreshed `core22` and `mesa-core22` which still works. I just tried refreshing those snaps on the new device, and `ubuntu-frame` runs at startup now, but only when hard resetting the device, but not after safely powering down/restarting 🤷🏻‍♂️

```
Name              Version               Rev    Tracking       Publisher    Notes
core22            20240111              1125   latest/stable  canonical**  base
mesa-core22       23.0.4                234    latest/stable  canonical**  -
pi-kernel         5.15.0-1053.56        821    22/stable      canonical**  kernel
ubuntu-frame      139-mir2.16.4+dev332  9217   22/edge        canonical**  -
```

jaydensmith, 2024-04-29:
From what I can see, the times `ubuntu-frame` won't start, the logs show `No EDID data available on connector 89` just before listing the DRM device details.

jaydensmith, 2024-04-30:
It appears a different brand screen resolved the issue. Strange that it was working before, but I'll consider it an anomaly.

closed 2024-04-30 by jaydensmith