canonical/ubuntu-frame #192: Ubuntu Frame not working on Core 24 on Raspberry Pi 5

jpm-canonical, 2024-06-28:
Starting with a clean Core 24 image made with Pi Imager. When you ssh into the device you get a motd with example commands:

```
In this image, why not create an IoT web-kiosk. First, connect a 
screen, then run: 

   snap install ubuntu-frame wpe-webkit-mir-kiosk
   snap set wpe-webkit-mir-kiosk url=https://ubuntu.com/core
```

Before running these commands I have a blank black screen actively being displayed on the monitor.

Executing the first of these two commands results in a black screen with a flashing text cursor in the top left hand corner.
```
$ snap install ubuntu-frame wpe-webkit-mir-kiosk
ubuntu-frame (22/stable) 144-mir2.17.0 from Canonical✓ installed
wpe-webkit-mir-kiosk (22/stable) 2.38.5 from GLANCR labs (glancr) installed
```
[core-24-frame-22-logs.txt](https://github.com/user-attachments/files/16025779/core-24-frame-22-logs.txt)

Further guidance suggest one should use the 24 channel of ubuntu-frame on core 24:
```
$ snap refresh --channel 24 ubuntu-frame
ubuntu-frame (24/stable) 152-mir2.17.0 from Canonical✓ refreshed
$ sudo reboot
```

This still only showed the black screen with flashing cursor. Log file from journalctl attached.
[core-24-frame-24-logs.txt](https://github.com/user-attachments/files/16025815/core-24-frame-24-logs.txt)

Further debugging suggested to check if the kms dtoverlay is enabled.

On the SD card in the `ubuntu-seed` partition, there is a config.txt file containing the lines:
```
# Enable the FKMS ("Fake" KMS) graphics overlay, and allocate 128Mb to the
# GPU memory
dtoverlay=vc4-fkms-v3d,cma-128
```

### Workaround 
Changing the last line from **fkms** to **kms** solves the problem.
```
dtoverlay=vc4-kms-v3d,cma-128
```

After startup the grey grandiented background is displayed and shortly replaced by the webkit kiosk browser.

Saviq, 2024-06-28:
I can't confirm that on my side, flashed the image that Pi Imager chooses for Pi5 / Core 24, and indeed it had `fkms` in there - but it worked just fine…

Would be good to check what `graphics-test-tools.drm-info` says when you flip it back to `fkms`?

jpm-canonical, 2024-06-28:
Before I revert the output of `graphics-test-tools.drm-info` is: 
[drm-info-kms.txt](https://github.com/user-attachments/files/16026916/drm-info-kms.txt)

After I revert to `fkms` I again get the black screen with flashing text cursor. 
Running `graphics-test-tools.drm-info` responds with:
```
jpm@localhost:~$ sudo graphics-test-tools.drm-info
drmGetDevices: No such file or directory
```

Saviq, 2024-06-28:
Well, that makes it clear it's not a Frame issue, but I wonder what's different with mine, that it's happy on both kms and fkms.

I can also confirm on the devices in our lab that using the below image things work out of the box… with the `fkms` overlay o.O

http://cdimage.ubuntu.com/ubuntu-core/24/dangerous-stable/pending/ubuntu-core-24-arm64+raspi.img.xz

```
ubuntu@localhost:~$ grep kms /run/mnt/ubuntu-seed/config.txt 
dtoverlay=vc4-fkms-v3d,cma-128

ubuntu@localhost:~$ uname  -a
Linux localhost 6.8.0-1006-raspi #6-Ubuntu SMP PREEMPT_DYNAMIC Mon Jun 17 11:30:51 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

ubuntu@localhost:~$ sudo snap install graphics-test-tools
graphics-test-tools (22/stable) 22.04 from Canonical✓ installed

ubuntu@localhost:~$ sudo graphics-test-tools.drm-info | head
drmModeGetResources: Operation not supported
Node: /dev/dri/card0
├───Driver: vc4 (Broadcom VC4 graphics) version 0.0.0 (20140616)
│   ├───DRM_CLIENT_CAP_STEREO_3D supported
│   ├───DRM_CLIENT_CAP_UNIVERSAL_PLANES supported
│   ├───DRM_CLIENT_CAP_ATOMIC supported
│   ├───DRM_CLIENT_CAP_ASPECT_RATIO supported
│   ├───DRM_CLIENT_CAP_WRITEBACK_CONNECTORS supported
│   ├───DRM_CAP_DUMB_BUFFER = 1
│   ├───DRM_CAP_VBLANK_HIGH_CRTC = 1
│   ├───DRM_CAP_DUMB_PREFERRED_DEPTH = 24
```

Only thing comes to mind is somehow your display hardware is making the Pi firmware unhappy with `fkms`?…

Either way, closing here as it's not a Frame issue.

closed 2024-06-28 by Saviq

jpm-canonical, 2024-07-16:
> Only thing comes to mind is somehow your display hardware is making the Pi firmware unhappy

Just to note here, I've tried booting the Pi 5 connected to a different monitor, also connected using the [micro-HDMI to HDMI](https://www.raspberrypi.com/products/micro-hdmi-to-standard-hdmi-a-cable/) cable. This display had the same effect.