canonical/ubuntu-frame #193: Wayland Socket Not Found Even After Successful Connection on Ubuntu Core 24 (Raspberry Pi 5)

Luciogc, 2024-07-17:
I'm trying to set up a web kiosk using `wpe-webkit-mir-kiosk` on Ubuntu Core 24 running on a Raspberry Pi 5. While the snap connection to Ubuntu Frame's Wayland interface appears to be established, the kiosk service fails to launch, continuously reporting that the Wayland socket is not available.

**Steps to Reproduce:**

1. Installed `wpe-webkit-mir-kiosk` and Ubuntu Frame on Ubuntu Core 24 on a Raspberry Pi 5.
2. Set the desired URL using `sudo snap set wpe-webkit-mir-kiosk url=<my-url>`.
3. Enabled the service to start automatically: `sudo snap set wpe-webkit-mir-kiosk daemon=true`
4. Connected the snap to Wayland: `sudo snap connect wpe-webkit-mir-kiosk:wayland` 
5. Started the service: `sudo snap start wpe-webkit-mir-kiosk`
6. Observed repeated errors in the logs:
    ```
    Wayland socket not available, waiting one second...
    ```

**Expected Behavior:**

The kiosk should start up and display the specified URL.

**Actual Behavior:**

The kiosk service fails to launch, and the logs show the error mentioned above.

**Additional Information:**

- `snap connections wpe-webkit-mir-kiosk` shows the Wayland connection is established.
- `snap logs ubuntu-frame` shows that Ubuntu Frame is running.
- `ss -lnp | grep wayland` shows no process listening on the Wayland socket.
- Relevant parts of the Ubuntu Frame logs:
  ```
  [... timestamp ...] eglstream: EGLDeviceEXTs found, but none are suitable for Mir
  [... timestamp ...] mirserver: (Unsupported by system environment)
  [... timestamp ...] ERROR: Failed to find any platforms for current system
  ```

**Environment:**

- Ubuntu Core version: 24
- wpe-webkit-mir-kiosk version: 2.38.5
- Ubuntu Frame version: 144-mir2.17.0
- Device: Raspberry Pi 5

**Possible Causes:**

- **Compatibility:** The version of `wpe-webkit-mir-kiosk` (2.38.5) might not be fully compatible with the newer version of Ubuntu Frame (144-mir2.17.0) on Ubuntu Core 24.
- **Raspberry Pi 5 Specific Issues:**  There might be specific issues related to the Raspberry Pi 5's graphics hardware or driver support in Ubuntu Frame.
- **Configuration:** Ubuntu Frame or `wpe-webkit-mir-kiosk` might be misconfigured.

**Potential Solutions (Attempted):**

- Rebooted the system
- Restarted both services
- Ran the `wpe-webkit-mir-kiosk` setup script (`sudo /snap/wpe-webkit-mir-kiosk/current/bin/setup.sh`)
- Checked and confirmed that no other processes are using the Wayland socket.

**Additional Considerations for Raspberry Pi 5:**

- **Driver Support:** Check if the latest firmware and graphics drivers for the Raspberry Pi 5 are installed.
- **Hardware Acceleration:** Ensure that hardware acceleration is enabled in Ubuntu Frame (if supported on the Pi 5).

I would greatly appreciate any assistance in resolving this issue, especially considering the Raspberry Pi 5 context. Please let me know if there are any other details I should include. Thanks!

Saviq, 2024-07-17:
@Luciogc thanks for the detailed report!

This looks suspiciously like #192, can you try changing `dtoverlay=` from `fkms` to `kms` in the `config.txt` file? I'll post the exact location in a few minutes.

Saviq, 2024-07-17:
FYI: we'll be fixing Core 24 images to have `kms` by default, as `fkms` is not supported any more. Unclear how it works on my setup :shrug:.

Saviq, 2024-07-17:
Please reopen if that doesn't help in your case.

closed 2024-07-17 by Saviq