canonical/ubuntu-frame #201: Allow privileged snaps to access sensitive protocols

Saviq, 2024-09-20:
There are valid reasons to allow external snaps to e.g. capture screenshots, or provide input methods.

It's currently only possible to globally enable those protocols, opening up for attacks.

We need a way to allow privileged (e.g. through snap interface connections) snaps to access those, and for the device operator to decide.

Applying apparmor labels is what we always thought can be the mediation layer here.

AlanGriffiths, 2024-09-20:
One thing we've talked about in the past is sourcing the `auth_model` content from a configuration file managed by a snap option.

Saviq, 2024-09-20:
> One thing we've talked about in the past is sourcing the `auth_model` content from a configuration file managed by a snap option.

IMO we should try and focus on snap interfaces, that gives a more integrated story (e.g. autoconnections, brand/dedicated store  assertions).