canonical/ubuntu-frame #209: Missing apparmor entries on Raspberry Pi

laxamar, 2024-12-09:
On a raspberry Pi (4/5) the following entries are missing from apparmor and are stopping deployment:
(files are in /var/lib/snapd/apparmor/profiles) for the daemon version

I add them as an #include in /etc/apparmor.d/local but it gets overwritten with every update: 
`# Grant read, write, and execute permissions to the /root/.cache/ directory itself
/root/.cache/ rw,

# Grant read and write permissions to all contents within /root/.cache/
/root/.cache/** rw,

/root/.cache/mesa_shader_cache/ rw,
/root/.cache/mesa_shader_cache/** rwk,
/root/.cache/mesa_shader_cache/index rwk,

/root/ r,

# Similarly, adjust other directories as needed
/run/user/0/ rw,
/run/user/0/** rwk,
/run/user/0/snap.wpe-webkit-mir-kiosk/cog-shared-* rw,

# Allow access to DRM resources
/dev/dri/** r,`

Can we add them permanently or have an easy way to add local?

laxamar, 2024-12-09:
For snap.wpe-webkit-mir-kiosk.daemon:

`/proc/zoneinfo r,
/proc/zoneinfo/** r,
capability sys_admin,
capability sys_nice,
/sys/fs/cgroup/system.slice/snap.webkit-mir-kiosk.daemon.service/** r,
/sys/fs/cgroup/system.slice/{,**/}snap.@{SNAP_INSTANCE_NAME}{,.*}/{memory.current,memory.max,memory.high} r,`

Saviq, 2024-12-10:
Hi @laxamar, the profiles are managed outside of the snap itself (in [SnapD interfaces](https://github.com/canonical/snapd/tree/master/interfaces)). Please suggest changes to interfaces there.

The Mesa shader cache shouldn't be a fatal problem, do you have the `vc4-kms-v3d` overlay enabled?

That said, we should probably direct it somewhere to make use of the shader cache. I'll file an issue with the Mesa snaps.

---

The WPE ones are debatable, whether they should be allowed (and/or in which interface). Something to discuss on the [Snapcraft forums](https://forum.snapcraft.io/) possibly.

--

I'll close this here as this isn't a anything to solve within Frame itself.

closed 2024-12-10 by Saviq