canonical/ubuntu-frame #210: X11 visual has unknown pixel format. bits_per_rgb_value: 8, red_mask: 63488

Bummbumm6, 2025-01-11:
Hello, I am trying to run Ubuntu Frame with a screen for the raspberry pi with the semi-official piscreen driver. When it starts automatically, it outputs correctly through HDMI, but not through the screen, and when forcing it to run on display 0 throws the error in the title. Here's a full log:
```
+ [ /snap/ubuntu-frame/11648/bin/run-frame = --help ]
+ mkdir -p /home/windigerbus/snap/ubuntu-frame/11648/.config/
+ grep -vE (^vt=|^console-provider=vt) /var/snap/ubuntu-frame/11648/frame.config
+ rm -f /home/windigerbus/snap/ubuntu-frame/11648/.config/frame.display
+ cp /var/snap/ubuntu-frame/11648/frame.display /home/windigerbus/snap/ubuntu-frame/11648/.config/
+ exec /snap/ubuntu-frame/11648/bin/run-frame /snap/ubuntu-frame/11648/bin/gpu-2404-wrapper /snap/ubuntu-frame/11648/usr/local/bin/frame --debug
+ [ /snap/ubuntu-frame/11648/bin/gpu-2404-wrapper = --help ]
+ dirname /run/user/1000/snap.ubuntu-frame
+ XDG_RUNTIME_DIR=/run/user/1000
+ export XDG_RUNTIME_DIR
+ mkdir -p /run/user/1000 -m 700
+ [ -n  ]
+ exec /snap/ubuntu-frame/11648/bin/gpu-2404-wrapper /snap/ubuntu-frame/11648/usr/local/bin/frame --debug
[2025-01-11 17:06:57.991686] < - debug - > miral: Loaded display configuration file: /home/windigerbus/snap/ubuntu-frame/11648/.config/frame.display
[2025-01-11 17:06:58.006222] <information> mirserver: Starting
[2025-01-11 17:06:58.006290] < - debug - > mirserver: Not trying logind: "DISPLAY" is set and X need not have claimed the VT
[2025-01-11 17:06:58.006378] < - debug - > mirserver: Not using Linux VT subsystem for session management: Failed to find the current VT
[2025-01-11 17:06:58.006411] < - debug - > mirserver: No session management supported
[2025-01-11 17:06:58.006433] <information> VT switch key handler: No VT switching support available: MinimalConsoleServices does not support VT switching
[2025-01-11 17:06:58.006599] <information> mircommon: Loading modules from: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform
[2025-01-11 17:06:58.006670] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/graphics-eglstream-kms.so.22
[2025-01-11 17:06:58.006687] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.22
[2025-01-11 17:06:58.006697] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/renderer-egl-generic.so.22
[2025-01-11 17:06:58.006707] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/server-x11.so.22
[2025-01-11 17:06:58.006716] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/input-evdev.so.10
[2025-01-11 17:06:58.008509] <information> mirserver: Found display driver: mir:x11 (version 2.19.2)
[2025-01-11 17:06:58.008976] <information> mirserver: Driver supports:
[2025-01-11 17:06:58.009057] <information> mirserver: 	System (priority 192)
[2025-01-11 17:06:58.009620] <information> mirserver: Found display driver: mir:egl-generic (version 2.19.2)
[2025-01-11 17:06:58.010273] <information> mirserver: Found display driver: mir:gbm-kms (version 2.19.2)
[2025-01-11 17:06:58.012416] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card0, driver v3d
[2025-01-11 17:06:58.039323] <information> gbm-kms: Failed to probe DRM device: ./src/platforms/common/server/kms-utils/drm_mode_resources.cpp(42): Throw in function mir::graphics::kms::DRMModeResUPtr {anonymous}::resources_for_drm_node(int)
Dynamic exception type: boost::wrapexcept<std::system_error>
std::exception::what: Couldn't get DRM resources: Operation not supported

[2025-01-11 17:06:58.039871] < - debug - > gbm-kms: Quirks: checking device with devnode: /dev/dri/card1, driver vc4-drm
[2025-01-11 17:06:58.040085] <information> gbm-kms: Failed to probe DRM device: ./src/server/console/minimal_console_services.cpp(134): Throw in function virtual std::future<std::unique_ptr<mir::Device> > mir::MinimalConsoleServices::acquire_device(int, int, std::unique_ptr<mir::Device::Observer>)
Dynamic exception type: boost::wrapexcept<std::system_error>
std::exception::what: Failed to acquire DRM master: Operation not permitted

[2025-01-11 17:06:58.040465] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver 
[2025-01-11 17:06:58.040642] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver 
[2025-01-11 17:06:58.040816] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver 
[2025-01-11 17:06:58.040991] < - debug - > gbm-kms: Quirks: checking device with devnode: (null), driver 
[2025-01-11 17:06:58.041103] <information> mirserver: Driver supports:
[2025-01-11 17:06:58.041180] <information> mirserver: 	/devices/platform/axi/1002000000.v3d/drm/card0 (priority 0)
[2025-01-11 17:06:58.041908] <information> mirserver: Found display driver: mir:eglstream-kms (version 2.19.2)
[2025-01-11 17:06:58.042721] < - debug - > eglstream: Found EGLDeviceEXT with device extensions: EGL_EXT_device_drm EGL_EXT_device_drm_render_node
[2025-01-11 17:06:58.043004] < -warning- > eglstream: Failed to check DRM modesetting support for device 1002000000.v3d: Invalid argument (22)
[2025-01-11 17:06:58.043501] < - debug - > eglstream: Found EGLDeviceEXT with device extensions: EGL_MESA_device_software EGL_EXT_device_drm_render_node
[2025-01-11 17:06:58.043563] <information> eglstream: Failed to query DRM node for EGLDevice: Failed to determine DRM device node path from EGLDevice: EGL_BAD_PARAMETER (0x300c)
[2025-01-11 17:06:58.043574] <information> mirserver: Driver supports:
[2025-01-11 17:06:58.043580] <information> mirserver: 	/devices/platform/axi/1002000000.v3d/drm/card0 (priority 0)
[2025-01-11 17:06:58.043598] <information> mirserver: Selected display driver: mir:x11 (version 2.19.2) for platform
[2025-01-11 17:06:58.043623] <information> EGL: EGL_KHR_debug logging enabled at maximum verbosity
libEGL warning: DRI2: failed to authenticate
[2025-01-11 17:06:58.085898] <information> mircommon: Loading modules from: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform
[2025-01-11 17:06:58.085997] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/graphics-eglstream-kms.so.22
[2025-01-11 17:06:58.086019] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/graphics-gbm-kms.so.22
[2025-01-11 17:06:58.086047] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/renderer-egl-generic.so.22
[2025-01-11 17:06:58.086066] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/server-x11.so.22
[2025-01-11 17:06:58.086079] <information> mircommon: Loading module: /snap/ubuntu-frame/11648/usr/lib/aarch64-linux-gnu/mir/server-platform/input-evdev.so.10
[2025-01-11 17:06:58.087249] <information> mirserver: Found rendering driver: mir:eglstream-kms (version 2.19.2)
[2025-01-11 17:06:58.087282] < - debug - > eglstream: No outputs capable of accepting EGLStream input detected
[2025-01-11 17:06:58.087293] < - debug - > eglstream: Probing will be skipped
[2025-01-11 17:06:58.087304] <information> mirserver: (Unsupported by system environment)
[2025-01-11 17:06:58.087317] <information> mirserver: Found rendering driver: mir:gbm-kms (version 2.19.2)
[2025-01-11 17:06:58.087329] < - debug - > gbm-kms: No outputs capable of accepting GBM input detected
[2025-01-11 17:06:58.087338] < - debug - > gbm-kms: Probing will be skipped
[2025-01-11 17:06:58.087347] <information> mirserver: (Unsupported by system environment)
[2025-01-11 17:06:58.087357] <information> mirserver: Found rendering driver: mir:egl-generic (version 2.19.2)
[2025-01-11 17:06:58.087369] <information> mirserver: Driver supports:
[2025-01-11 17:06:58.087378] <information> mirserver: 	System (priority 192)
[2025-01-11 17:06:58.087393] <information> mirserver: Found rendering driver: mir:x11 (version 2.19.2)
[2025-01-11 17:06:58.087462] <information> mirserver: Selected rendering driver: mir:egl-generic (version 2.19.2) for platform
[2025-01-11 17:06:58.115834] <information> platform-generic-egl: Cannot enable linux-dmabuf import support: EGL_EXT_image_dma_buf_import_modifiers not supported
[... cut ...] tModifiers::EXTImageDmaBufImportModifiers(EGLDisplay)
Dynamic exception type: boost::wrapexcept<std::runtime_error>
std::exception::what: EGL_EXT_image_dma_buf_import_modifiers not supported

ERROR: ./src/platforms/x11/x11_resources.cpp(60): Throw in function MirPixelFormat {anonymous}::pixel_format_for_visual(const xcb_visualtype_t&)
Dynamic exception type: boost::wrapexcept<std::runtime_error>
std::exception::what: X11 visual has unknown pixel format. bits_per_rgb_value: 8, red_mask: 63488
```

AlanGriffiths, 2025-01-13:
> the semi-official piscreen driver

This is probably the source of your problem. Linux GPU drivers come in two parts, a kernel module and a corresponding userspace driver. Frame gets its userspace driver from a content interface (`gpu-2404`) and the default provider of that is `mesa-2404` which provides the widely used Mesa drivers.

 You would need to connect an alternative content snap to provides the userspace drivers for the "semi-official piscreen driver" for this to work. (It isn't hard to package alternative drivers, but I'm not aware of such a snap, not of the licencing implications of distributing it)

Saviq, 2025-01-13:
@Bummbumm6 this shows that your graphics driver does not have the extensions required by Mir:

```
std::exception::what: EGL_EXT_image_dma_buf_import_modifiers not supported
```

Ref.: https://canonical-mir.readthedocs-hosted.com/stable/explanation/mir-graphics-support/#gbm-kms

Saviq, 2025-01-13:
@RAOF can you comment on whether there's a way out?

RAOF, 2025-01-13:
As documented there, `EGL_EXT_image_dma_buf_import_modifiers` is optional; Mir *should* work fine without it (although it's somewhat difficult to test that, as all common drivers *do* support that extension).

I think the error preventing this from working is 
```
ERROR: ./src/platforms/x11/x11_resources.cpp(60): Throw in function MirPixelFormat {anonymous}::pixel_format_for_visual(const xcb_visualtype_t&)
Dynamic exception type: boost::wrapexcept<std::runtime_error>
std::exception::what: X11 visual has unknown pixel format. bits_per_rgb_value: 8, red_mask: 63488
```
from the X11 platform; from above in the log
```
[2025-01-11 17:06:58.043598] <information> mirserver: Selected display driver: mir:x11 (version 2.19.2) for platform
```
it looks like ubuntu-frame is running nested in an X11 server. Is that the expected behaviour for the piscreen? If so, it looks like that X11 environment is unusual in a way that our X11 backend is not expecting.

RAOF, 2025-01-13:
Poking around a bit more, a `red_mask` of `63488` is the top 5 bits of a 16-bit value. That suggests the underlying X server is using a 16-bit 5/6/5 colour format, which we don't support.

Bummbumm6, 2025-01-14:
> Poking around a bit more, a `red_mask` of `63488` is the top 5 bits of a 16-bit value. That suggests the underlying X server is using a 16-bit 5/6/5 colour format, which we don't support.

What color formats do you support? How do I change it?

> it looks like ubuntu-frame is running nested in an X11 server. Is that the expected behaviour for the piscreen?

piscreen takes display input from /dev/fb1, and I knew I could use an X server and set it to directly output there with fbdev, so I though that would be easiest. I guess that doesn't work, so is it possible to output frame directly to that framebuffer?

> Linux GPU drivers come in two parts, a kernel module and a corresponding userspace driver. Frame gets its userspace driver from a content interface (gpu-2404) and the default provider of that is mesa-2404 which provides the widely used Mesa drivers.

For piscreen I am loading the kernel overlay piscreen.dtb that's included in Raspberry Pi OS, but not anything directly in the userspace. I guess that is not needed due to piscreen outputting to screen from a framebuffer?

I am sorry for asking these questions, I'm a newbie.