canonical/ubuntu-frame #220: `applications:` need to think they're fullscreen still

Saviq, 2025-04-24:
Steps:
- configure Frame to place a snap/app custom
  https://mir-server.io/docs/how-to-configure-ubuntu-frame-for-multiple-outputs#heading--custom-client-placement
- start the configured app

Expected:
- no window decorations

Current:
- decorations (e.g. scummv)

[image]

mattkae, 2025-04-24:
Would the solution be to set the "always use csd" flag?

Saviq, 2025-04-24:
> Would the solution be to set the "always use csd" flag?

This is what Frame does to other apps:
https://github.com/canonical/ubuntu-frame/blob/67680d581deeae47af2a7b4496204685d2b367cf/src/frame_window_manager.cpp#L73-L78

The app might be told its fullscreen, to hide chrome et al, but not be _really_ fullscreen ;)

I suppose the ultimate right thing here would be to have a way to tell the app it's tiled…

AlanGriffiths, 2025-04-25:
xdg-shell v5 has the `tiled_*` states (which we don't currently provide) and v7 also has `constrained_*` - that's the information we _should_ be supplying

mattkae, 2025-04-28:
I _think_ the short-term solution here is to always use CSD.

I think the long term solution is to provide tiled states (although who knows which clients actually respect these states). `constrained_*` also makes sense. (Side note: I want all of this in miracle!)

mattkae, 2025-04-30:
From all of the clients that I've tested, the `tiled_*` states are not synonymous with min-fullscreens. The title bar will always be there, for example. Some times drop shadow is removed though.