canonical/ubuntu-frame #221: Apps going fullscreen despite custom size

Saviq, 2025-04-24:
Steps:
- configure Frame to place mir-kiosk-neverputt custom
  https://mir-server.io/docs/how-to-configure-ubuntu-frame-for-multiple-outputs#heading--custom-client-placement
   ```yaml
   layouts:
     default:
       cards:
       - card-id: 0

       applications:
       - snap-name: mir-kiosk-neverputt
         position: [640, 0]
         size: [640, 512]
   ```
- start frame and neverputt

Expected:
- placed at the configured layout

Current:
- decorations shown at the current layout (#220)
- app shown in fullscreen size, positioned according to layout

[image]

mattkae, 2025-04-24:
Well this could be the case that the client starts as fullscreen, but it disrespects our request :)

The option here _might_ be to set the clip area on surfaces that disrespect our requests

AlanGriffiths, 2025-04-25:
> it disrespects our request to no longer be fullscreen, which is valid

Is it? The client makes _requests_, the server **tells** the clients of _events_.

PR #222 (2025-04-30): (#221) windows are now clipped to their tile if they are absolutely positioned and sized
fixes #221 

## Problem
- Some clients will not abide by our wishes for them to be a particular size

## Solution
- Even if the buffer is too large, we set the clip area accordingly so that the client at least does not overflow outside of the tile
commit 55d3ea31 (2025-04-30): (#221) windows are now clipped to their tile if they are absolutely positioned and sized (#222)

fixes #221

## Problem
- Some clients will not abide by our wishes for them to be a particular
size

## Solution
- Even if the buffer is too large, we set the clip area accordingly so
that the client at least does not overflow outside of the tile
closed 2025-04-30 by mattkae